Legion2/CorsairLightingProtocol #2: Make iCUE detect the Arduino correctly

Legion2, 2019-01-12:
Link 4 detects and displays the Arduino (Lighting Node PRO) in the UI.
In iCUE it does not show up. Only the USB disconnected sound is heard and the USB activity LED is flashing on start up of iCUE. No HID Reports are received by RawHID.
The following is logged by iCUE:
```
cue.dev: Device vid=1b1c pid=c0b ignored by HidDeviceEnumerator
```

Legion2, 2019-01-14:
The Arduino(Lighting Node PRO) VID=1b1c PID=0c0b is not detected/ignored by iCUE, because iCUE only recognize the Lighting Node Loader VID=1b1c PID=0c0c. So with only Lighting Node PRO (PID=0c0b) USB device nothing shows up.
With the Arduino being an Lighting Node Loader(VID=1b1c PID=0c0c) iCUE shows a LNP in the settings UI, but you can't select a model.
[image]

willianacp, 2019-07-11:
Eu estou com o mesmo problema, estou usando uma versão 0.4.0. How do I solve it?

willianacp, 2019-07-11:
I'm having the same problem, I'm using a version 0.4.0. How do I solve it?

Legion2, 2019-07-11:
do you use Arduino IDE or Visual Studio?

willianacp, 2019-07-11:
Visual Studio

Legion2, 2019-07-11:
Please open a new issue with the [tech-support template](https://github.com/Legion2/CorsairLightingProtocol/issues/new?assignees=&labels=&template=tech-support.md&title= ) and provide more information about your hardware and software settings.

willianacp, 2019-07-11:
Okay, if you can add me on discord.  TheGameTown#7653

Legion2, 2019-07-11:
@willianazambuja I send you invite

chanyork, 2021-09-22:
I need help on this as well.