Legion2/CorsairLightingProtocol #6: 5050 Leds

Bchap9, 2019-02-09:
Is it possible for me to use this to control 4 pin 5050 leds while still using the corsair protocol?

Legion2, 2019-02-09:
5050 leds aren't addressable. You can't control each led individual.

It's possible to set the color of the 5050 leds, by syncing it with one led of a 10 leds Lighting Strip from Corsair protocol.

Legion2, 2019-02-09:
```
#include <CorsairLightingProtocol.h>
#include <LEDController.h>
#include <FastLED.h>

#define NUM_LEDS CHANNEL_LED_COUNT * 2

#define RED_PIN 3
#define GREEN_PIN 4
#define BLUE_PIN 5

CRGB leds[NUM_LEDS];

void setup() {
#ifdef DEBUG
	Serial.begin(115200);
#endif
	pinMode(RED_PIN, OUTPUT);
	pinMode(GREEN_PIN, OUTPUT);
	pinMode(BLUE_PIN, OUTPUT);

	LEDController.addLeds(0, leds);
	LEDController.addLeds(1, &(leds[CHANNEL_LED_COUNT]));
	CorsairLightingProtocol.begin();
}

void loop() {
	if (CorsairLightingProtocol.available())
	{
		Command command;
		CorsairLightingProtocol.getCommand(command);
		CorsairLightingProtocol.handleCommand(command);
	}

	if (LEDController.updateLEDs()) {
		set4PinLEDs(leds[0]);
	}
}

void set4PinLEDs(const CRGB& color) {
	analogWrite(RED_PIN, color.r);
	analogWrite(GREEN_PIN, color.g);
	analogWrite(BLUE_PIN, color.b);
}
```
this should work (not tested).

closed 2019-03-02 by Legion2

willianacp, 2019-07-10:
is giving this error.

LEDController.h:20: In file included from
LEDController.cpp:16: from
FastLED.h:17: note  #pragma message  FastLED version 3.002.000
   #    pragma message "FastLED version 3.002.000"
 
LEDController.h:20: In file included from
Sketch4.ino:2: from
FastLED.h:17: note  #pragma message  FastLED version 3.002.000
   #    pragma message "FastLED version 3.002.000"
 
Sketch4.ino: 5:18: error: 'CHANNEL_LED_COUNT' was not declared in this scope
   #define NUM_LEDS CHANNEL_LED_COUNT * 2
Sketch4.ino:13: note  in expansion of macro NUM_LEDS
   void setup() {
 
Sketch4.ino: In function void setup()
 
Sketch4.ino: 23:14: error: missing template arguments before '.' token
   CorsairLightingProtocol.begin()
 
Error compiling project sources
Sketch4.ino: 23:26: error: 'leds' was not declared in this scope
   CorsairLightingProtocol.begin()
Debug build failed for project 'Sketch4'
 
Sketch4.ino: 24:14: error: missing template arguments before '.' token
 
Sketch4.ino: 24:33: error: 'CHANNEL_LED_COUNT' was not declared in this scope
 
Sketch4.ino: 25:24: error: expected unqualified-id before '.' token
 
Sketch4.ino: In function void loop()
 
Sketch4.ino: 29:28: error: expected primary-expression before '.' token
 
Sketch4.ino: 32:24: error: expected unqualified-id before '.' token
 
Sketch4.ino: 33:24: error: expected unqualified-id before '.' token
 
Sketch4.ino: 36:18: error: missing template arguments before '.' token
 
Sketch4.ino: 37:13: error: 'leds' was not declared in this scope

Legion2, 2019-07-10:
Yes, I updated the library, so the sketch must also be updated. This should compile with the version v0.4.0.
```
#include <CorsairLightingNodePRO.h>
#include <FastLED.h>

#define RED_PIN 3
#define GREEN_PIN 4
#define BLUE_PIN 5

// The number of LEDs per channel, there are two channels.
// 60 LEDs per channel is the maximum iCUE can handle.
#define CHANNEL_LED_COUNT 10

// Total count of LEDs on all channels, the value is calculated based on the leds per channel.
#define NUM_LEDS (CHANNEL_LED_COUNT * 2)

LEDController<CHANNEL_LED_COUNT> ledController(true);
CorsairLightingProtocol cLP(&ledController, firmware_version);

// This array conatins all RGB values for all LEDs of the both channels.
CRGB leds[NUM_LEDS];

void setup() {
#ifdef DEBUG
	Serial.begin(115200);
#endif
	pinMode(RED_PIN, OUTPUT);
	pinMode(GREEN_PIN, OUTPUT);
	pinMode(BLUE_PIN, OUTPUT);

	ledController.addLeds(0, leds);
	ledController.addLeds(1, &(leds[CHANNEL_LED_COUNT]));
	cLP.begin();
}

void loop() {
	if (cLP.available())
	{
		Command command;
		cLP.getCommand(command);
		cLP.handleCommand(command);
	}

	if (ledController.updateLEDs()) {
		set4PinLEDs(leds[0]);
	}
}

void set4PinLEDs(const CRGB& color) {
	analogWrite(RED_PIN, color.r);
	analogWrite(GREEN_PIN, color.g);
	analogWrite(BLUE_PIN, color.b);
}
```

willianazambuja, 2019-07-11:
the more it hangs and the led from the Arduino RX gets on.