Legion2/CorsairLightingProtocol #8: Explain how to use the library with ArduinoIDE

Legion2, 2019-03-09:


insuf, 2019-03-15:
sorry to bother you is there any chance i can run it on arduino uno r3 i can modify the board txt but then i get errors for RAWHID unxpected ' and bla bla can't find the lib for HID and so on?

Legion2, 2019-03-15:
The only way to use it on an arduino uno is by using https://github.com/NicoHood/HoodLoader2 bootloader.
But that will limit the capabilities in terms of memory and I have not tested it.
I recommend you to read the HoodLoader2 Readme, it's interesting to know that the Arduino has two chips that you can program.
You can also do the compile check, to test if some board works with the library, without having the hardware.

What did you mean with you modified the borad.txt, did you use ArduinoIDE?

insuf, 2019-03-15:
i used arduino IDE but to be fair i will do your method!

insuf, 2019-03-15:
HoodLoader2 gives me a lot of errors im new to arduino i can manage to program rainbow effects and customize the led strip ws2811 but i want to sync to my ICUE witch will have cooler effect.

insuf, 2019-03-15:
i just don't use VS and don't have license

Legion2, 2019-03-15:
I recommend you to use a PRO Micro board, it't not an official Arduino board but can be programmed using arduino. It's cheap and supports all required features.

For Visual Studio, there is a Community Edition, which should also be sufficient.

insuf, 2019-03-15:
Thank you very much i will find i way around it its a second board that i buy i had intel galileo not compatible with fastled because i got it from shop pre owned for 5£ and then i read that is not supported by fastled this is my second board. I just found out for your topic now i would of consider to buy PRO micro but to late i already buy full setup for it i plan to put everything under my desk if don't find a way i will just stick with fastLED it's perfect just not sync to the mouse and keyboard setup  witch is fine because even my mobo is asus so no icue aswell :)

Legion2, 2019-03-15:
For sync iCUE, Asus and others there is a tool [JackNet RGB Sync](https://fanman03.ml/?page=dwn), it's not perfect, but it works.
You can use this [Arduino Sketch](https://github.com/DarthAffe/RGB.NET/blob/master/RGB.NET.Devices.WS281X/Sketches/RGB.NET_Arduino.ino) for Arduino Uno to sync with the tool. I don't know how well it works.
For questions about these there are two discord servers [JackNet RGB Sync](https://discordapp.com/invite/Gr6f9XZ) and [RGB.NET](https://discord.gg/9kytURv).

SamPoulton, 2019-03-15:
@Legion2 if it's helpful, I've been trying to use it with the 16u2 on my Mega with HL2 but it seems significantly too big to fit.

Legion2, 2019-03-18:
@SamPoulton I worked on support for Mega using HL2 #9. On the dev branch there are two new example sketches which are working on Arduino Mega with HL2. Using the 16u2 only as bridge to reduce the memory requirement to a minimum.

Legion2, 2019-05-26:
added Wiki page [How to use Arduino IDE](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Arduino-IDE)

commit 45cc3e88 (2019-06-02): updated Requirements

added How to use Arduino IDE wiki close #8
closed 2019-06-08 by Legion2