Legion2/CorsairLightingProtocol #11: Cannot add multiple Node PRO

omerorhan, 2019-04-08:
I have one `Lighting Node PRO` already connected and working nice. But when i add one more both don`t work.

Legion2, 2019-04-09:
Is the first `Lighting Node PRO` a official one? When i connect two fake LNP at least one is working. I think that has something to do with the deviceID or the Serial Number are the same on all Arduinos. Currently the deviceID is [hard coded](https://github.com/Legion2/CorsairLightingProtocol/blob/master/src/CorsairLightingFirmware.h#L40), you can try to change it for one LNP to see if it works. The Serial Number is not working at all `(I/O Status 31 Ein an das System angeschlossenes Gerät funktioniert nicht.)`. [Here](https://github.com/Legion2/CorsairLightingProtocol/blob/master/src/RawHID.cpp#L62) you can enter a Serial Number but for me it don't work.

omerorhan, 2019-04-09:
Two of them are fake. I will try to change Serial No. Also I tried to change one of them to "Commander Pro". Windows shows it connected. But icue don't even recognise CP in the device list.

Legion2, 2019-04-09:
Commander Pro has an extended command set that is not implemented, so when iCUE try to use these commands an error code is returned.

Legion2, 2019-04-09:
Serial Number and DeviceID must be different, then iCUE show both.

Legion2, 2019-04-09:
Don't use `00 00 00 00` as DeviceID, this will generate a random one.

omerorhan, 2019-04-10:
I tried ,

device ID
`uint8_t DeviceId[4] = { 00, 00, 00, 01 };`

SERIAL_NUMBER
`#ifndef SERIAL_NUMBER
#define SERIAL_NUMBER "1EA000060904BAAEFB66DF55421900F8"
#endif`

But still same.

Legion2, 2019-04-10:
Did you use the latest master branch, with the fix?

Legion2, 2019-04-10:
Verify in device Manager (Windows) that the serial number is set

omerorhan, 2019-04-10:
yes, i checkout the latest master branch. I will check.

omerorhan, 2019-04-10:
i`ve tried with your the latest commit.

I am using 2 ARDUINO. 1 LEONARDO and 1 Pro Micro.
Both are listed as` Lighting Node Pro` at devices menu in Windows 10.
When I check within USBDeview Serial Numbers setted OK.
Device ID`s seems OK with DeviceIDTool.

But now, None of them are recognised in iCUE.

LEONARDO
DEVICE CODE : 55 03 EE 09
SERIAL : 3DG000070805CFFEFC77DF55856300A5

 Pro Micro
DEVICE CODE : 50 04 FF 20
SERIAL : 7DH000070805CFFEFC77DF55856300E7

iCUE Ver : 3.14.104

Legion2, 2019-04-11:
[image] for me it's working.(the numbering comes form the tests with different ids)
You have two options:
1. DEBUG mode: add `-DDEBUG` in the board.txt in the same line where the -DSERIAL... is
in the code apply the diff [debug.diff.txt](https://github.com/Legion2/CorsairLightingProtocol/files/3068025/debug.diff.txt), upload and open Serial Monitor.

2. Use [system information viewer](http://rh-software.com/) it supports the Corsair protocol on the PC side. The tool is very complex but really helps debugging.
`[USB Bus]` shows information of all USB devices(Serial Number)
`[Status |▼]`->`[Link Status]` shows information on Protocol layer
it's best to read the "AIO Device Control" section on the website

omerorhan, 2019-04-13:
I tried lots of things.

With old commits when SERIAL NUMBER is `Status 31 A device attached to the system is not functioning` they work separately. But not together.

With the latest commit, no single unit is working.

Below my results, i hope they are useful.

[image]

[image]

Legion2, 2019-04-13:
Please use the "Release" [configuration](https://github.com/MicrosoftDocs/visualstudio-docs/blob/master/docs/debugger/how-to-set-debug-and-release-configurations.md#change-the-build-configuration) in Visual Studio.
It looks to me that the Serial Number is a hex number, so only characters 0-9 and A-F are allowed.
If you use the debug flag you always have to connect a Serial Monitor, otherwise the Arduino does not work properly.

The `Command: 6B` is strange, it does not exist.

Legion2, 2019-04-13:
Please make as few changes as possible from the example, so change the `CHANNEL_LED_COUNT` to 50 and revert all other code changes. Only change one digit of the Serial Number and use a simple DeviceID like `00 00 00 01` so its easy to reproduce your setup. Than make sure the single Arduino works with the latest master branch and setup hardwareplayback and close iCUE, so the Arduino is in hardwareplayback mode and the settings are saved in the EEPROM. Restart the iCUE Service, so the disconnected devices are not shown in the UI, this may require a PC restart because of a bug in iCUE. Do the same with the second Arduino. After that `unknown led channel mode` should no longer be logged to the Serial Monitor.

omerorhan, 2019-04-14:
i tried exactly what have you said. I gave my hours but still same. If there is serial number registered it just dont work. Only works single node when it says `Status 31 A device attached to the system is not functioning`. Maybe the problem is somewhere else on my system I don`t know.

omerorhan, 2019-04-14:
Can`t you make that SERIAL NUMBER and DeviceID thing auto generated?

Legion2, 2019-04-15:
The DeviceID can be auto generated.
Set it to `00 00 00 00`, iCUE will generate a new one next time it connects to the LNP.

Legion2, 2019-04-15:
@omerorhan did you use the `-DDEBUG` flag when getting `Status 31 A device attached to the system is not functioning`?

omerorhan, 2019-04-16:
1- I delete everthing under AppData\Local\Temp.
2- I change the Serial Number and set -DDEBUG on.
3- Start From VS2019(VS2017) at Release Mode.
2- Open Serial Monitor From Arduino IDE.
3- Restart From ICUE Settings.

And it`s working. I saw the serial number is OK.

Then i restart computer.
Repeat the process again without -DDEBUG.
When I restart ICUE nothing happens. It just don`t work.

When I try to change DEVICE ID, i realize that its always 00 00 00. It dont change.

Legion2, 2019-04-16:
single Arduino or both?

omerorhan, 2019-04-16:
single

Legion2, 2019-04-16:
I think there is a memory leak, buffer overflow or a timing issue somewhere in the library.

omerorhan, 2019-04-16:
I left Serial Number empty. Only Change DeviceId.
They both works now :)

Legion2, 2019-04-16:
WTF!? OK, leave it that way.

Legion2, 2019-04-29:
@omerorhan I now have fixed several problems and reduced the length of the Serial Number. I tested with two Pro Micro with Serial Numbers `FB66DF55421900F5` and `FB66DF55421900F6`, DeviceIDs `4E 91 67 B9` and `52 42 79 85`, without `-DDEBUG` flag and in release mode and having no problems. Please can you try and report any problems.

omerorhan, 2019-05-03:
@Legion2 i tried and it is working without problem. Thank you.

closed 2019-05-04 by Legion2