Legion2/CorsairLightingProtocol #12: Arduino Micro Driver Error

Lethegrin, 2019-04-18:
Receiving a driver error after compiling the code for the arduino micro [bought here on amazon.
](https://www.amazon.com/gp/product/B012FOV17O/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

Perhaps I don't have the correct drivers installed? I have no other corsair devices.

Legion2, 2019-04-18:
Which board you compiled the code for?
You get the error when you try to upload or after the upload?

Lethegrin, 2019-04-19:
Fixed. Again, I am using the Pro Micro [bought here on amazon.](https://www.amazon.com/gp/product/B012FOV17O/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

First I added the board to the board manager by following [this tutorial.](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/installing-windows)

Afterwards I chose the correct board in the dropdown list. (in my case it was the SparkFun Pro Micro 5V 16 MHz). 

Next, I flashed the board with the example sketch "SimpleLightingCOntroller.ino".

After the example sketch was flashed to my board, I was still receiving a driver error so I uninstalled the device completely by going into Settings>Bluetooth & other devices>(left click)Lighting Node Pro>Remove device.

Finally I unplugged the usb connecting the Pro Micro and reconnected it. After of which it showed up in iCUE. Success.

Thanks for the help.

closed 2019-04-19 by Legion2