Legion2/CorsairLightingProtocol #15: CLP For Corsair MM800 Clone

blurry-mind, 2019-05-16:
Hello Legion2, I really like your CLP. Is it possible to make the leds controlled using CLP as a Corsair MM800 RGB Mousepad?
I have succesfully managed to make the arduino detected as MM800 RGB in Device, but it doesnt detected in iCUE.
Thanks.

Legion2, 2019-05-16:
I don't know the protocol that MM800 RGB is using, why do you what the CLP to show up as MM800 RGB? This library implements the functionality of a Lighting Node PRO, so changing the name will not change the functionality.

Legion2, 2019-05-16:
[From this project](https://github.com/ckb-next/ckb-next/blob/master/src/daemon/led_mousepad.c), it looks like the protocol of mouse pads are different from the CLP.

blurry-mind, 2019-05-17:
> 
> 
> I don't know the protocol that MM800 RGB is using, why do you what the CLP to show up as MM800 RGB? This library implements the functionality of a Lighting Node PRO, so changing the name will not change the functionality.

I'm just curious if there is any possibility to build a mouse pad that can be controlled as a fake MM800 using iCUE. What i have done is changing the pid according to the MM800 pid, but it showed many errors (some null object) on iCUE log file. Anyway, thanks for your quick response. I will build an RGB mouse pad based on your LNP

Legion2, 2019-05-18:
I will close the Issue for now.  If you have any question on how to use this library in your project, you can ask them here.

closed 2019-05-18 by Legion2