Legion2/CorsairLightingProtocol #17: Not showing up as Corsair Lightning node pro

Powertx, 2019-05-21:
I have a chinese pro micro with sparkfun bootloader. I upload the simple LightningNodePro sketch, and it doesnt show up as a LightningNodePro in windows... 
I am using arduino IDE.

Powertx, 2019-05-21:
Ok. I got it to show in device manager as LightningNode pro. And it shows up in ICue. But the device manager says missing driver.

Legion2, 2019-05-21:
Try the solution from #12.
Which Windows version do you use?
How did you upload the sketch with the Arduino IDE or did you use Visual Studio to get it show up in iCUE?

Powertx, 2019-05-21:
I have windows 7. I used arduino ide. I mannualy changed the vid and pid for the sparkfunboard and burnt a bootloader with those values. Removing LightningNodePro from device manager didnt help.

Legion2, 2019-05-21:
Maybe a problem with your custom bootloader, did you [install the driver](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/installing-windows) for windows 7?

Powertx, 2019-05-21:
I did install the drivers. I will try burning the default sparkfun bootloader tommorow,

Legion2, 2019-05-21:
Try to get a basic arduino example running on your pro micro, then try the LightningNodePro sketch.

Powertx, 2019-05-22:
With the default bootloader, my pro micro doesnt show up as a lighning node pro. Arduino examples work so all is good there.

Powertx, 2019-05-22:
I uploaded the code with visual studio. Then it shows up as LightningNodePro with default bootloader.
But under device manager it still shows up under unknown devices.

Legion2, 2019-05-22:
- Where does it show up as LightningNodePro, is iCUE detecting it, does it work in iCUE, are there more details about the error like a error code from windows?
- What ist the url of the package.json and the exact name of the board used to upload the sketch(some thing like `https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json`)?
- Did you use the [latest version](https://github.com/Legion2/CorsairLightingProtocol/releases/latest) of this library, because you are talking about "simple LightningNodePro" sketch, but this was renamed to "SingleStripLightingNodePRO" 18 days ago.

Powertx, 2019-05-27:
1.iCue is detecting it. It shows up in device manager as LightningNodePro, it might be just that windows7 thinks its an unknown device. Windows 10 didnt say anything about unknown device.

2.https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json
Sparkfun Pro Micro Atmega32U4 5v 16MHz

3.I did use the the latest version. I am using the sketch LightningNodePRO

Also it works better when i use visual studio.
For me it doesnt work when i upload the skecth with arduino ide. Since to my knowladge it doesnt change the VID,PID and manufacturer values.

Sorry for late response, i didnt have time to mess with arduino.

Powertx, 2019-06-02:
Ok. Windows 7 shows under it "unknown devices" in device manager. But it still works with iCUE.
So all is good.

closed 2019-06-02 by Legion2