Legion2/CorsairLightingProtocol #20: Support Commander PRO Features

Legion2, 2019-05-28:
Make it possible to create a  Commander PRO with this library, so the the protocol is fully implemented.
Features that a Commander PRO have in addition to that of a LNP:
- [ ] temperature sensors
- [ ] voltage measurement
- [ ] fan control
- [ ] fan speed sensor

mountaser, 2019-05-28:
that would be dope i was actually going to suggest it!

Legion2, 2019-06-08:
@mountaser do you have hardware to test #21? Temperature- and Fan-Controller must be tested.

TylerSeiford, 2019-06-08:
@Legion2 I have a pro micro, what exactly do you want tested?

I don't believe I have a method for temperature input to the arduino but I can test fan outputs.

Legion2, 2019-06-08:
You can not power the fans directly from the arduino, the arduino only provides the pwm signal, so you need some wiring to use the arduino as fan controller.
For temperature all sensors that are compatible with most mainboards temperature connectors are supported, these sensors are NTC Thermistor 10 KOhms. They require a 10 KOhms resistor in series and then you measure the voltage between them.

A Commander PRO is more complex than a LNP.

mountaser, 2019-06-08:
@Legion2  i can test fan control speed i have ML fans i can use them to test 
Don't have any temp sensors tho

Legion2, 2019-06-08:
Ok, currently only fixed % power is supported by the fan controller. I will also implement fan curve support, but without measuring the fan speed.

mountaser, 2019-06-08:
@Legion2 i can test anything for u 
Currently not at home spending time with some family but i can when i get back
Do u have telegram discord or anyway of communicating? 
My telegram is @mountaser_halak and my discord is mountaser #8924

Legion2, 2019-06-08:
@mountaser Discord: Legion#8743

Legion2, 2019-06-21:
PR is merged, this feature is in the next release.

PR #21 (2019-06-21): Support Commander PRO Features
Make it possible to create a Commander PRO with this library, so the the protocol is fully implemented close #20. This pull request add full support of the Protocol to the the library, but does not implement the functionality of a CommanderPRO completely. You can implement your own fan controller based on the `FanController` base class. A simple implemetnation can be found in the `CommanderPRO` example.
Features that a are supported by the example sketch `CommanderPRO`:
* read temperature sensors
* ~~voltage measurement~~
* fan control fixed power
* fan control c [... cut]
closed 2019-06-21 by Legion2