Legion2/CorsairLightingProtocol #25: Would like to get this project working with the arduino nano, what are the limitations which prevent support for this ?

jamesgohl, 2019-06-19:
**Describe the problem**
A clear and concise description of your problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino Leonardo]
 - IDE: [e.g. Arduino IDE]
 - Version: [e.g. 0.3.0]
 - Sketch: [e.g. SingleStripLightingNodePRO]

**Code changes**
Did you changed the code of the library or the sketch?

Legion2, 2019-06-19:
The Arduino Nano is based on an **ATmega328** and **ATmega16U2**, yes two microcontroller on one board. The ATmega328 does not handle the USB communication on the Arduino Nano, the ATmega16U2 handle the communication. The sketch you upload, is uploaded to the  ATmega328, so you can't change what the ATmega16U2 is doing. But that is important for this library. This library needs control over the USB communication in such a way that 1. VID and PID can be changed 2. new USB Endpoint can be defined and 3. HID Reports can be send and received. Nothing of that can be done with the Arduino Nano, because you can not program the ATmega16U2.

The **ATmega32u4** for example has built-in USB communication. Boards like Arduino Leonardo and Pro Micro use this microcontroller and because you can program the microcontroller that handle the USB communication, you can do point 1-3 from above.

Maybe #9 helps you, but as far as I know does Hoodloader2 not work with arduino nano.

closed 2019-06-19 by Legion2

jamesgohl, 2019-06-19:
Thank you for the explanation, it makes perfect sense.  So if I buy a Arduino Micro ATmega32U4 this should work then.  What happens once the VID and PID are changed on the board? Is it no longer possible to upload code to from the Arduino IDE?

Legion2, 2019-06-19:
It is still possible to upload code, it only has another COM port. By the way you should use Visual Studio to upload this sketch or read [this](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Arduino-IDE) for Arduino IDE.

jamesgohl, 2019-06-19:
> 
> 
> It is still possible to upload code, it only has another COM port. By the way you should use Visual Studio to upload this sketch or read [this](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Arduino-IDE) for Arduino IDE.

Sounds good, shouldn't be a problem following these instructions.  You have saved me a whole world of time not having to write my own controller software in C# to drive my corsair fan :)

Will this work? Because it mentions having to choose "Arduino Leonardo" from the Board menu?...
https://www.amazon.co.uk/gp/product/B07FQBQ4Z6/ref=ox_sc_act_title_2?smid=A1F4U6XVWUBG1U&psc=1

Legion2, 2019-06-19:
Yes I use those too.

> Sounds good, shouldn't be a problem following these instructions. You have saved me a whole world of time not having to write my own controller software in C# to drive my corsair fan :)

Just to be noted here: This library is not for driving corsair fans by an arduino, it's for using LED strips that can be controlled with [FastLED](http://fastled.io/). I don't know if the corsair fan use the same protocol as normat LED strips, as I don't own corsair fans.
> Will this work? Because it mentions having to choose "Arduino Leonardo" from the Board menu?...

Where did i wrote "Arduino Leonardo" should be chosen?

TylerSeiford, 2019-06-20:
I'd assume that the corsair fans are using some sort of common LED control protocol but that's not something I have sitting around and can try to figure out.

jamesgohl, 2019-06-20:
> Just to be noted here: This library is not for driving corsair fans by an arduino, it's for using LED strips that can be controlled with [FastLED](http://fastled.io/). I don't know if the corsair fan use the same protocol as normal LED strips, as I don't own corsair fans.

Yes I am very well aware of what this project consists of and what is possible with it, thank you for letting me know however.  I should have chosen my words more carefully, I mean the lighting strips which are inside the corsair fans.
So I can let you know that the corsair fans have LED strips in them, WS2812 to be exact and I have the fan LEDs working with FastLED alone.

> Where did i wrote "Arduino Leonardo" should be chosen?

Okay so following the URL I included, links you to the product: Arduino Micro. It states "Please click the [Tool] and choose the module [Arduino Leonardo] when you open the Arduino IDE software." on the product description.

TylerSeiford, 2019-06-20:
> So I can let you know that the corsair fans have LED strips in them, WS2812 to be exact and I have the fan LEDs working with FastLED alone.

That's actually awesome to know, now if I ever end up getting some I can do some custom control on them!

> "Arduino Leonardo" 
It doesn't actually matter if you choose the micro or leonardo (I've used both by mistake), they both use the same microcontroller chip and similar enough pinouts to not be an issue.  Most of them say to use the leonardo because I think it has a slightly different enough pinout to matter.