Legion2/CorsairLightingProtocol #27: Support for fan tach signal

jamesgohl, 2019-06-25:


Legion2, 2019-06-25:
Currently fan tacho signal is explicitly not supported, because of the implementation complexity.
It would require to use hardware interrupts, because of the timing critical measurements.
But hardware [interrupts are disabled](https://github.com/FastLED/FastLED/wiki/Interrupt-problems) while FastLED updates the leds. So no tacho signal can't be measured all the time.

jamesgohl, 2019-06-25:
> Currently fan tacho signal is explicitly not supported, because of the implementation complexity.
> It would require to use hardware interrupts, because of the timing critical measurements.
> But [...]

This does not matter since I can build a frequency to voltage converter, and plug this into analog pin.

Legion2, 2019-06-26:
That's the problem with the Commander PRO Support, there is not the one way - like FastLED for **ALL** leds - there are many solutions to the same problem and this library should only handle the protocol and USB handling. The `LEDController` is part of the library, because it is sufficient in 99% of all use cases. I created base classes for the Commander PRO features, so everyone can implement their solution without the need to handle USB and protocol stuff.
By design you don't need to modify any code of the library to get your custom controller up and running. You only have to implement the abstract base classes and instantiate a `CorsairLightingProtocol` like in the [CommanderPRO.ino](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/CommanderPRO/CommanderPRO.ino).

**TL;DR** The support of custom solutions is out of scope of this library and will not be implemented here.

closed 2019-06-30 by Legion2