Legion2/CorsairLightingProtocol #29: ICUE does not identify.

willianacp, 2019-07-11:
Was using version 0.3.0, was working everything right identified in the application. I ended up getting a new arduino, I went to the version 0.4.0 in the new arduino and without wanting to finish what was working. now it does not identify anymore in the icue, if a mute the PID = 0c0b for PID = 0c0c it appears only that it is the same as the image. and in the new arduino the same [thing.](http://prntscr.com/odr5vg)

Legion2, 2019-07-11:
Which arduino, iCUE version, OS?
Do you have to arduinos connected at the same time?
Which sketch did you used?
Why you not using the Issue Tech-support template?

PS: you can include images natively in GitHub no need for file hoster

willianazambuja, 2019-07-11:
ICUE version: 3.17.94. Windows 10.
only one connected arduino.
[image]

Legion2, 2019-07-11:
`cLNP.begin();` is missing in the `setup` method.

Legion2, 2019-07-11:
https://github.com/Legion2/CorsairLightingProtocol/blob/v0.4.0/examples/LightingNodePRO/LightingNodePRO.ino#L31 this is the example sketch for v0.4.0, you are using the example sketch of `dev` (v0.5.0).

Legion2, 2019-07-13:
now v0.5.0 is release so, this sketch should work with the library from the library manager.

closed 2019-07-13 by Legion2