Legion2/CorsairLightingProtocol #30: Unable to flash Uno Mega controller

adam120800, 2019-07-11:
**Describe the bug**
HoodLoader2UnoMegaController project wont compile. I am not much programming savvy and i use Arduino IDE always so i am a little lost when it comes properly reporting. I will attach as many screenshots as possible :D. I have included all libraries.

[image]
[image]
[image]
[image]

**System (please complete the following information):**
 - OS: Win 10
 - Board: Legit Arduino Uno with HoodLoader2 properly installed on 16u2
 - IDE: Visual Studio 2019
 - Version: Latest commit, downloaded as zip
 - Sketch: HoodLoader2UnoMegaController.ino

**Code changes**
None

Legion2, 2019-07-12:
First screenshot looks like you have installed an older version of the library too. Use the library Manager to remove it and install the downloaded zip as library

Legion2, 2019-07-12:
Left of the green play button change "Debug" to "Release" like [this](https://github.com/MicrosoftDocs/visualstudio-docs/blob/master/docs/debugger/how-to-set-debug-and-release-configurations.md#change-the-build-configuration).

Legion2, 2019-07-12:
I will release the version with Arduino Mega support soon, then you only have to update the library in the library manager.

adam120800, 2019-07-12:
Awesome, thanks. I ,managed to manually update the library. Uploaded uno mega controller and clpbridge successfully to both main chip and 16u2. My arduino is recognized as lightining node pro, TX and RX are blinking but i have no output to my led's whatsoever

adam120800, 2019-07-12:
Also i have no errors while compiling or anything similiar

Legion2, 2019-07-12:
Debug the HoodLoader2CLPBridge sketch, add -DDEBUG in the board.txt of that sketch folder, upload HoodLoader2CLPBridge again and open serial monitor with rate 1000000

adam120800, 2019-07-12:
Okay, i will do that tonight i am going to work currently

closed 2019-08-28 by stale[bot]