Legion2/CorsairLightingProtocol #32: Lighting node pro sp120 fans not working

Arctic445, 2019-07-17:
I'm having trouble getting my Chinese arduino (atmega32u4) to work.
I've gotten to the point where iCue recognizes it, but its not working
I've set it up for 6 sp120 fans but they are not lighting up.

I've connected the data wire to pin 2 and ground to ground.
[image]

I'm using the example code for the lighting node pro because i'm rather new to coding

Arctic445, 2019-07-17:
I'm using arduino IDE 1.8.9 because I couldn't get visual studio to work.

I'm using this board:

promicroclp.name=Arduino Leonardo
leonardo.vid.0=0x2341
promicroclp.pid.0=0x0036
promicroclp.vid.1=0x2341
promicroclp.pid.1=0x8036
promicroclp.vid.2=0x2A03
promicroclp.pid.2=0x0036
promicroclp.vid.3=0x2A03
promicroclp.pid.3=0x8036

promicroclp.upload.tool=avrdude
promicroclp.upload.protocol=avr109
promicroclp.upload.maximum_size=28672
promicroclp.upload.maximum_data_size=2560
promicroclp.upload.speed=57600
promicroclp.upload.disable_flushing=true
promicroclp.upload.use_1200bps_touch=true
promicroclp.upload.wait_for_upload_port=true

promicroclp.bootloader.tool=avrdude
promicroclp.bootloader.low_fuses=0xff
promicroclp.bootloader.high_fuses=0xd8
promicroclp.bootloader.extended_fuses=0xcb
promicroclp.bootloader.file=caterina/Caterina-Leonardo.hex
promicroclp.bootloader.unlock_bits=0x3F
promicroclp.bootloader.lock_bits=0x2F

promicroclp.build.mcu=atmega32u4
promicroclp.build.f_cpu=16000000L
promicroclp.build.vid=0x1b1c
promicroclp.build.pid=0x0c0b
promicroclp.build.usb_product="Lighting Node PRO"
promicroclp.build.usb_manufacturer="Corsair"
promicroclp.build.board=AVR_LEONARDO
promicroclp.build.core=arduino
promicroclp.build.variant=leonardo
promicroclp.build.extra_flags={build.usb_flags} '-DSERIAL_NUMBER="FB66DF55421900F5"'

And this code:

#include <CorsairLightingNodePRO.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[CHANNEL_LED_COUNT];
CRGB ledsChannel2[CHANNEL_LED_COUNT];

CorsairLightingNodePRO cLNP(ledsChannel1, ledsChannel2);

void setup() {
  disableBuildInLEDs();
  FastLED.addLeds<NEOPIXEL, DATA_PIN_CHANNEL_1>(ledsChannel1, CHANNEL_LED_COUNT);
  FastLED.addLeds<NEOPIXEL, DATA_PIN_CHANNEL_2>(ledsChannel2, CHANNEL_LED_COUNT);
}

void loop() {
  cLNP.update();
}

Legion2, 2019-07-17:
SP120 uses a special protocol, I don't know which, without this information you can't control them.

Arctic445, 2019-07-17:
I thought it was UCS1903

Legion2, 2019-07-17:
then you have to change FastLED.addLeds... to use that protocol

Arctic445, 2019-07-17:
Its working! thanks for the help!

Legion2, 2019-07-17:
keep in mind to not use the arduino as power supply

closed 2019-07-17 by Legion2

Arctic445, 2019-07-17:
I'm using the sata power in the hub. And the 5v is not connected to the arduino.

Arctic445, 2019-07-17:
Now I have the issue that the color is incorrect. Red is blue, Green is red and blue is green.

I've checked the fastled page for a fix but the code is not compiling

void setup() { 
 FastLED.addLeds<UCS1903,DATA_PIN_CHANNEL_1, BRG>(leds, NUM_LEDS 6);
}

Error:
exit status 1
'leds' was not declared in this scope

Legion2, 2019-07-17:
there is a missing `,` between `NUM_LEDS`  and `6`

Arctic445, 2019-07-17:
that's not working either but I got it fixed lol

FastLED.addLeds<UCS1903,DATA_PIN_CHANNEL_1, BRG>(ledsChannel1, CHANNEL_LED_COUNT);

Legion2, 2019-07-17:
good, I added the protocol information to the [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Corsair-fans-or-other-hardware)