Legion2/CorsairLightingProtocol #33: lightnodepro not installing as hid i think

rikis156, 2019-07-21:
**Describe the problem**
i have the same problem as #17 
it did install correctly the first time but it did not work so i uninstalled it and re-plugged my arduino and now it just says unknown device under com ports

**Screenshots**
[image]
[image]

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Pro Micro
 - IDE: Arduino IDE
 - Version: 0.3.0 all the way to 0.5.0
 - Sketch:  LightingNodePRO

**Code changes**
Did you changed the code of the library or the sketch?
no i did not edit anythning

rikis156, 2019-07-21:
################################################################################
################################## Pro Micro ###################################
################################################################################
promicrocpl.name=CLP SparkFun Pro Micro

promicrocpl.upload.tool=avrdude
promicrocpl.upload.protocol=avr109
promicrocpl.upload.maximum_size=28672
promicrocpl.upload.maximum_data_size=2560
promicrocpl.upload.speed=57600
promicrocpl.upload.disable_flushing=true
promicrocpl.upload.use_1200bps_touch=true
promicrocpl.upload.wait_for_upload_port=true

promicrocpl.bootloader.tool=avrdude
promicrocpl.bootloader.unlock_bits=0x3F
promicrocpl.bootloader.lock_bits=0x2F
promicrocpl.bootloader.low_fuses=0xFF
promicrocpl.bootloader.high_fuses=0xD8

promicrocpl.build.board=AVR_PROMICRO
promicrocpl.build.core=arduino:arduino
promicrocpl.build.variant=promicro
promicrocpl.build.mcu=atmega32u4
promicrocpl.build.usb_product="Lighting Node PRO"
promicrocpl.build.vid=0x1b1c
promicrocpl.build.extra_flags={build.usb_flags}

######################### Pro Micro 3.3V / 8MHz ################################
promicrocpl.menu.cpu.8MHzatmega32U4=ATmega32U4 (3.3V, 8 MHz)

promicrocpl.menu.cpu.8MHzatmega32U4.build.pid.0=0x0c0b
promicrocpl.menu.cpu.8MHzatmega32U4.build.pid.1=0x0c0b
promicrocpl.menu.cpu.8MHzatmega32U4.build.pid=0x0c0b
promicrocpl.menu.cpu.8MHzatmega32U4.build.f_cpu=8000000L

promicrocpl.menu.cpu.8MHzatmega32U4.bootloader.extended_fuses=0xFE
promicrocpl.menu.cpu.8MHzatmega32U4.bootloader.file=caterina/Caterina-promicro8.hex

############################# Pro Micro 5V / 16MHz #############################
promicrocpl.menu.cpu.16MHzatmega32U4=ATmega32U4 (5V, 16 MHz)

promicrocpl.menu.cpu.16MHzatmega32U4.build.pid.0=0x0c0b
promicrocpl.menu.cpu.16MHzatmega32U4.build.pid.1=0x0c0b
promicrocpl.menu.cpu.16MHzatmega32U4.build.pid=0x0c0b
promicrocpl.menu.cpu.16MHzatmega32U4.build.f_cpu=16000000L

promicrocpl.menu.cpu.16MHzatmega32U4.bootloader.extended_fuses=0xCB
promicrocpl.menu.cpu.16MHzatmega32U4.bootloader.file=caterina/Caterina-promicro16.hex
promicrocpl.build.extra_flags={build.usb_flags} '-DSERIAL_NUMBER="FB66DF55421900F5"'

################################################################################

if it may be of use :)

Legion2, 2019-07-21:
I don't understand what you mean with unknown COM Port. After you uploaded the sketch, the COM port changes, so if you what to upload again, you have to select the new COM port in Arduino IDE.

The HID is set like you can see in SIV, there is only the manufacturer unknown, because Pro Micro does not set it. You can add `promicroclp.build.usb_manufacturer="Corsair"` to the `boards.local.txt`, that should fix that.

rikis156, 2019-07-21:
i have added that but still dont show up in icue and last night i did get it to show up as an hid but it still wont work in icue

Legion2, 2019-07-21:
You have SIV, so go in SIV to LINK->LINK Status there should be CLNP, click on it and post screenshot here

rikis156, 2019-07-21:
[image]

i have two CLNP showing and both say the same

Legion2, 2019-07-21:
did you tried to restart your PC?

Legion2, 2019-07-21:
You can also try the solution from #12

rikis156, 2019-07-21:
i restarted my pc and got a different screen in SIV but it still did not show up in iCUE and yes i have tried #12

rikis156, 2019-07-21:
[image]
it looks like this now

Legion2, 2019-07-21:
use v0.5.0, comment the line `disableBuildInLEDs();` in the sketch and upload.
Then you should see the RX and TX led of the board blinking.

rikis156, 2019-07-21:
i have a yellow light that is flashing

Legion2, 2019-07-21:
i don't known if it's the problem, but maybe the EEPROM of arduino contains bad data. To test this use the SingleStripLightingNodePRO.ino and change `USE_EEPROM true` to false

rikis156, 2019-07-21:
no change  only the yellow light that do anything

Legion2, 2019-07-21:
how did you installed this library, are you sure it's v0.5.0?

rikis156, 2019-07-21:
drag and drop to libraries in the arduino IDE folder

rikis156, 2019-07-21:
[image]
not sure if this says anything but it dose not show up under hid in device manager as it did before

Legion2, 2019-07-21:
upload `DeviceIDTool.ino` to set a Device ID with serial monitor

rikis156, 2019-07-21:
Ok i did that but still nothing, rightnow only if i restart my pc it shows as a hid i the device manager if i reupload to the arduino it goes away

Legion2, 2019-07-21:
use different usb port?

Legion2, 2019-07-21:
use SIV to get the USB and HID inforamtion(in USB Bus click on the Port button of the LNP)

Legion2, 2019-07-21:
is see you called the id in `boards.local.txt` `promicrocpl` and not `promicroclp`, maybe there is somewhere else a typo.

rikis156, 2019-07-21:
not sure where that would be, i have not written anytning else than boards.local.txt and i did fix that promicroclp insted of promicrocpl but it acts like it has always done

Legion2, 2019-07-21:
after you changed boards.local.txt you must restart Arduino IDE

Legion2, 2019-07-21:
In SIV it should look something like this
[image]
[image]

rikis156, 2019-07-21:
[image]

rikis156, 2019-07-21:
the only thing i can see that is different is Device USB V2.00  LangID 0409 1 Class 0.0.0 Class 0 (00) SubClass 0 (00) Protocol 0 (00)  Maximum Packet Size 64 (EP0)  Configurations 1  Descriptors 14

Legion2, 2019-07-21:
you can add some debug code to the `loop` and see if it's printed to the serial monitor

rikis156, 2019-07-21:
it does :(

Legion2, 2019-07-21:
so you have serial output every time the loop gets called, but iCUE and SIV Link Status does not show good output?

rikis156, 2019-07-21:
[image]
yes it seems so

Legion2, 2019-07-21:
you are using an old Arduino IDE version, can you update?

Legion2, 2019-07-21:
Later this day i will create a debug sketch, that will give us more information.

rikis156, 2019-07-21:
i will update

rikis156, 2019-07-25:
In file included from C:\Program Files (x86)\Arduino\libraries\CorsairLightingProtocol-dev\src/FastLEDController.h:19:0,

                 from C:\Program Files (x86)\Arduino\libraries\CorsairLightingProtocol-dev\src/CorsairLightingNodePRO.h:20,

                 from C:\Program Files (x86)\Arduino\libraries\CorsairLightingProtocol-dev\examples\DebugSketch\DebugSketch.ino:16:

C:\Program Files (x86)\Arduino\libraries\FastLED-master/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.000

 #    pragma message "FastLED version 3.002.000"

                     ^

DebugSketch:31:18: error: 'PRINT_LOOP' was not declared in this scope

 bool printLoop = PRINT_LOOP;

                  ^

DebugSketch:32:20: error: 'PRINT_UPDATE' was not declared in this scope

 bool printUpdate = PRINT_UPDATE;

                    ^

Multiple libraries were found for "FastLED.h"
 Used: C:\Program Files (x86)\Arduino\libraries\FastLED-master
 Not used: D:\Dropbox\Arduino\libraries\arduino_883197
exit status 1
'PRINT_LOOP' was not declared in this scope

i get this error while i try to compile

Legion2, 2019-07-25:
you use the Arduino IDE, so you must define all the build options from the board.txt in the boards.local.txt. the DebugSketch require more build options then the normal sketch, add this extra options to you boards.local.txt.

rikis156, 2019-07-26:
for some reason when i uploded the DebugSketch.ino it all started to work for me 

and to anyone else that is using a pro micro this is the bord.text that made it work for me

################################################################################
################################## Pro Micro ###################################
################################################################################
promicroclp.name=CLP SparkFun Pro Micro

promicroclp.upload.tool=avrdude
promicroclp.upload.protocol=avr109
promicroclp.upload.maximum_size=28672
promicroclp.upload.maximum_data_size=2560
promicroclp.upload.speed=57600
promicroclp.upload.disable_flushing=true
promicroclp.upload.use_1200bps_touch=true
promicroclp.upload.wait_for_upload_port=true

promicroclp.bootloader.tool=avrdude
promicroclp.bootloader.unlock_bits=0x3F
promicroclp.bootloader.lock_bits=0x2F
promicroclp.bootloader.low_fuses=0xFF
promicroclp.bootloader.high_fuses=0xD8

promicroclp.build.board=AVR_PROMICRO
promicroclp.build.core=arduino:arduino
promicroclp.build.variant=promicro
promicroclp.build.mcu=atmega32u4
promicroclp.build.usb_product="Lighting Node PRO"
promicroclp.build.usb_manufacturer="Corsair"
promicroclp.build.vid=0x1b1c
promicroclp.build.extra_flags={build.usb_flags} '-DSERIAL_NUMBER="FB66DF55421900F5"'

######################### Pro Micro 3.3V / 8MHz ################################
promicroclp.menu.cpu.8MHzatmega32U4=ATmega32U4 (3.3V, 8 MHz)

promicroclp.menu.cpu.8MHzatmega32U4.build.pid.0=0x0c0b
promicroclp.menu.cpu.8MHzatmega32U4.build.pid.1=0x0c0b
promicroclp.menu.cpu.8MHzatmega32U4.build.pid=0x0c0b
promicroclp.menu.cpu.8MHzatmega32U4.build.f_cpu=8000000L

promicroclp.menu.cpu.8MHzatmega32U4.bootloader.extended_fuses=0xFE
promicroclp.menu.cpu.8MHzatmega32U4.bootloader.file=caterina/Caterina-promicro8.hex

############################# Pro Micro 5V / 16MHz #############################
promicroclp.menu.cpu.16MHzatmega32U4=ATmega32U4 (5V, 16 MHz)

promicroclp.menu.cpu.16MHzatmega32U4.build.pid.0=0x0c0b
promicroclp.menu.cpu.16MHzatmega32U4.build.pid.1=0x0c0b
promicroclp.menu.cpu.16MHzatmega32U4.build.pid=0x0c0b
promicroclp.menu.cpu.16MHzatmega32U4.build.f_cpu=16000000L

promicroclp.menu.cpu.16MHzatmega32U4.bootloader.extended_fuses=0xCB
promicroclp.menu.cpu.16MHzatmega32U4.bootloader.file=caterina/Caterina-promicro16.hex
promicroclp.build.build.extra_flags={build.usb_flags} '-DSERIAL_NUMBER="FB66DF55421900F5"'

################################################################################

 And Legion2  you the legend man thanks for all the help you have provided me :)

Legion2, 2019-07-26:
But you are now using the normal sketch like before? You uploaded the DebugSketch successfully and then switch back to the other sketch, or not?

rikis156, 2019-07-26:
iam not using the DebugSketch any more but i have some flicker in the light but that could be my connections but it works

Legion2, 2019-07-26:
I'm asking, because i want to understand the problem, which now is somehow fixed. If I understand it i can improve this project so others don't run into the same problem.

closed 2019-07-26 by Legion2