Legion2/CorsairLightingProtocol #41: Hardware related (Custom)

McHauge, 2019-10-02:
Hey, this is not much of a problem I'm having, nor is it directly a feature.

But I'm thinking about making a custom PCB with 2 controller chip onboard to utilize both USB's in the USB front header on the motherboard, for a total of 4 LED strip channels, on one board with one connection and one Molex power port in.

For this I'm wondering how the fans would map out in terms of port usage on the device, are they on their own individual io port or do they map into the same two channels, or maybe something else?

Thinking of using two ATmega8U2 for the MCU's, but I don't know if you got some info about if they are directly compatible, else I'll just use the bigger ATmega32U4.

Anything else I should look at what might be useful in such a board, basically two Pros in one package.

-Andreas

TylerSeiford, 2019-10-03:
Hey there, I've been working on a similar type project for a while [over here](https://github.com/TylerSeiford/CorsairArduinoController).

So far I've gotten the first round of prototype PCBs back and am currently using one to run the LED strip in this computer (non-addressable), the ARGB fans (coolermaster 200mm)'s RGB, and the same ARGB fan's ARGB.

My design (current prototype in hand) uses an Arduino Pro Micro (ATmega32u4) and has one 12v RGB header, two ARGB channels (each has a 12v and a 5v header), along with two fan headers.  I'm limited by the number of PWM channels available on the pro micro (5), and so I have been working on making another design with 5 fans and 4 temperature inputs.

As to the control, you can create up to 6 fan objects for iCUE to control, and how you map them to the physical fans is up to you (you could have 12 headers, two per channel if you want).  In addition, you have two addressable LED channels, which can have as many LEDs as iCUE allows (60 per channel off the top of my head).  You can map these to different physical ports [as shown here](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/LightingNodePRO/LightingNodePRO.ino) or to a single physical port [as shown here](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/SingleStripLightingNodePRO/SingleStripLightingNodePRO.ino).  These restrictions are per USB device/MCU, and so with two MCUs you could double these numbers.

McHauge, 2019-10-03:
Nice, I'm thinking to basically integrate the Pro micro onto the board directly, thereby also giving me access to all 6 PWM's there's a 6 on D13 (pin: 32) on the ATmega32u4.

On the PCB you have created now, you got 2 J-ARGB 1 and 2, just to be sure, are they for the ARGB fans and on the 3 pins is that the data return for lopping into the next RGB fan, from what I read the pins would have been swapped around, but I don't have one to test with my self, so just asking to be sure.

And on the temp side, do you need the temp sensor inputs or du you actually use it? With the dual temp sensor in the example, it might make sense to at least include them

McHauge, 2019-10-03:
This is how I'm planning to use the I/O on the ATmega32u4, anything I'm missing?
[image]

Might also make a pin-compatible to this layout with only the 2 LED channels, for an easy 
iCue -> USB -> LED 
control for behind a desk or something.

Legion2, 2019-10-05:
ATmega8U2 seems to have not enough memory to store all the led values and other runtime data.
I recommend not to use Molex for power supply but rather SATA-Power, because of the crappy Molex adapters that are not so easy to plug in and out.

TylerSeiford, 2019-10-05:
> J-ARGB 1 and 2

J-ARGB1 and J-ARGB2 are the 5V ARGB headers (like you would find on a motherboard), J-A1 & J-A2 are 12v ARGB, which is compatible with far more hobby style strips.

I don't have a way to chain the fans together in hardware, since these specific ones only come with a single ARGB connector that you can't chain together, but you can still chain multiple physical outputs to a single iCUE chain in your arduino firmware.

McHauge, 2019-10-07:
@Legion2 if you want to use the SATA-Power instead, where do you then buy the connector, or will you just cut a splitter up and use that? the Molex one just seemed easier to find as a connector. but will probably switch over to the SATA yes.

Also, I had a play around, and the only thing I had laying around with an ATmega32u4 was an electronic load project from earlier this year, but it worked. and is in the progress of getting a small single processer 2CH setup up and running. (have it ready for testing when I come home)

Added a picture of the crude test setup and the very rough test pcb:
[image]

[image]

[image]

McHauge, 2019-10-08:
A little progress update about the PCB, since the rough prototype worked, I went ahead and made a complete layout at: [https://easyeda.com/McHauge/open-corsair-lightning-node-pro-2ch-led-only](url)
And if you go there you can order it directly together with all the components needed for it, basically a kit.

Price for 10 complete sets is around 5-10$ for PCB and 60-80$ for the components in total. 
If you have something I have overlooked, or you think might need changing feel free to let me know.

I hope I'm not stepping over anyone's toes with this project.

Here are some pictures of the Layout:
[image]
[image]
[image]
[image]

McHauge, 2019-10-08:
just ordered 10pc for myself, with both PCB and components, about 7-8$ per unit before vat, after danish vat around 10-12$ for one unit including PCB, components and shipment.

McHauge, 2019-10-30:
Finally, the parts arrived and after mounting them up, all 10 got bootloaders (with the same serial) and set a device ID from "FF FF FF 01" up to "FF FF FF 0B" (01-10) via the example scetch. 
After this, all 10 got loaded with the default scetch for the Lighting node pro with pin 4 and 12 as LED outputs, and all 10 are working wonders, even tried 5 at once to one pc.
All got detected individually and could be controlled separately. 

All the files are available and open so if someone else could use one ([https://github.com/McHauge/OpenCorsairLighting](url)), you are free to order them your self, a minimum of 10 is recommended then by soldering it yourself it ends up at around 10$ per unit.

Some pictures of the final V1.0 product:
[image]
[image]
[image]
[image]

TylerSeiford, 2019-10-31:
Nice!

Legion2, 2019-11-14:
I'm closing this Issue, but you can continue discussing here.

closed 2019-11-14 by Legion2