Legion2/CorsairLightingProtocol #42: Arduino is not detected as LNP on first use

Legion2, 2019-10-05:
**Describe the bug**
After uploading an example sketch to the arduino, iCUE does not detect it.

**To Reproduce**
Steps to reproduce the behavior:
1. User example sketch without changes
2. Upload the sketch
3. See iCUE shows nothing

**Expected behavior**
iCUE detects the Arduino as LNP and LEDs can be controlled.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: PRO Mirco
 - IDE: Visual Studio
 - Version: 0.5.0
 - Sketch: LightingNodePRO

**Code changes**
None

**Related**
- #39 Icue dont show my "Lighting node pro"
- #33 lightnodepro not installing as hid i think
- #29 ICUE does not identify.

Legion2, 2019-10-05:
This is maybe related to the use of EEPROM to save and load data. On first use the data is loaded form not initialized EEPROM memory and contain wrong data. This maybe result in wrong behavior or broken runtime state of the program, from which it is not possible to recover.

McHauge, 2019-10-07:
Worked for me no problem last Friday, was just the boards.txt file I had to set up

Legion2, 2019-10-25:
This is fixed with Version 0.6.1 and 0.6.2

closed 2019-10-25 by Legion2