Legion2/CorsairLightingProtocol #58: Add Support for Corsair Lighting Node CORE

Legion2, 2019-11-14:
Make it possible to create a Corsair Lighting Node CORE with this library.
Needed information:
- [ ] USB PID
- [ ] USB Product Name
- [ ] Supported commands

Legion2, 2019-11-19:
I found the ids, there are documented in the [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Corsair-fans-or-other-hardware#rgb-controller).

Gannit, 2019-11-29:
What would the pinouts be for the node core? There are 6 ports vs the pro's 2 which would be very useful for multiple fans.  
Oh and thanks for the great work.

Legion2, 2019-11-29:
If you know the type of fans used, you can program the arduino to expose the 6 different pins for each fan, but else the Node CORE has one channel with one pin and you have to create a circuit that chains the fans. The chaining of fans must pass the output data signal of the previous to the next fan.

PR #79 (2019-12-05): Added example for Lighting Node CORE
close #58
closed 2019-12-05 by Legion2