Legion2/CorsairLightingProtocol #62: Step by step guide

hidefandy, 2019-11-17:
Hi,

I would really appreciate a step by step guide especially overcoming the led limits and maybe add  Spec Omega lighting node - 30 led Spec Omega case strip?

I created a guide on YouTube but it is probably not that accurate. I won’t link it here, but it can be  found under “Arduino ARGB Corsair Lighting Protocol Corsair Lighting Node Pro Corsair iCUE”. 

Ps. I have linked your project in the video as I think this is a great project.

Legion2, 2019-11-17:
Hi,
I also really want to do a step by step guide, the problem is there are so many configuration possibilities and hardware and software requirements that have to be considered for the guide. So I can't do only one guide for all this.
I splitted it up, so the main guide is in the Readme, it only applies to the ATmega32U4 and Visual Studio. For other options like using [Arduino IDE](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Arduino-IDE) or [Arduino Uno/Mega](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-on-Arduino-Uno-and-Arduino-Mega) I added guides in the Wiki.
I also know, using Visual Studio with the plugin is difficult to setup as non developer. That's all because of the Arduino IDE does not support to set the USB ID from the code. To overcome this I opened #57 to improve the setup for common boards like Arduino Uno, Mega, Leonardo and SparkFun PRO Micro.
These guides only cover the basic setup. For additional features, like the repeating or scaling of LED channels I created a separate [guide](https://github.com/Legion2/CorsairLightingProtocol#repeat-or-scale-led-channel).
The important point of this project is, that it is a library with some usage examples. It's designed to be very flexible and extensible and open for other developers. It's not a Plug and Play product and for many additional features you must have a bit of programming experience to use them.

The most work I'm currently doing in this project is to improve usability and provide many examples and templates so it's easy to understand how the features work. But I can't provide examples for all combination of features, boards and IDEs.

The Wiki is open for everybody, so if you find something unclear and want to add something you can edit the wiki. You can even create new guides that cover the whole setup for a specific configuration.

Legion2, 2019-11-17:
Do you have a Spec Omega case? The Spec Omega case have a  Lighting Node PRO and the 30 led Spec strip should be connected to it. I think the led stip is represented as 3 x 10 led strips in iCUE and can be controlled like any WS2812B strip. There is nothing special when using this library to control the 30 led Spec strip.

hidefandy, 2019-11-18:
Hi,

I appreciate the response as I am just an end user with an electrical background.

I understand the audience would generally be more acquainted.

No I don’t have the case however it does show up a single 30 led string In iCUE  which would be handy as it is 1M of standard 2812’s or 500mm of high density 60l/pm.

Probably has a specific vid and pid profile.

See 56 sec Mark.

https://www.youtube.com/watch?v=hod1WktB4oE

Legion2, 2019-11-19:
OK to support the 30 led strip, we need the USB PID of the controller of the Spec Omega case.

Legion2, 2019-11-19:
I opened issue #66 to track the process of getting that ids.

Legion2, 2019-11-20:
I created a [new repository](https://github.com/Legion2/CorsairLightingProtocolBoards) with Arduino boards definitions. The repo contains a package index json which can be added in the Arduino IDE and then you can install common boards like Pro Micro, Arduino Leonardo and HoodLoader 2 (Uno/Mega) directly in the Arduino IDE. All these boards have the correct USB IDs defined so they can be used to create the different corsair devices. The steps described in the [How to use Arduino IDE Wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Arduino-IDE) are not required anymore.

PR #70 (2019-11-23): Added a step by step guide in Readme
replaced Visual Studio with Arduino IDE in Readme
Added a step by step guide close #62
Better Arduino IDE support using the CLP Boards close #57
Updated Arduino IDE [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/Add-new-board-for-Arduino-IDE)
Added How to develop with Visual Studio [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-develop-with-Visual-Studio)
Updated Arduino Uno and Arduino Mega [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-on-Arduino-Uno-and-Arduino-Mega)

TODO:
- [x] Check all [... cut]
commit c8d8133d (2019-11-23): replaced Visual Studio with Arduino IDE in Readme
Added a step by step guide close #62
Better Arduino IDE support using the CLP Boards close #57
closed 2019-11-23 by Legion2