Legion2/CorsairLightingProtocol #65: Corsair iCUE LS100 Smart Lighting Strip support

Comango104, 2019-11-19:
Assuming it uses a ATmega32U4, what would you require to make this possible?
Would love to be able to make a cheap version of this controller.

McHauge, 2019-11-19:
I made this hardware and have used it in my pc for about a month now, the total cost per unit was around 7-10% (bought 10 at once since that's the lowest that makes sense), with both PCB and components ordered directly from china in one go. It's schematic is based around the Arduino Leonardo.

It's also using the ATmega32u4 and the software from here, so with a quick software update and it should be good to go. 

[OpenCorsairLighting](https://github.com/McHauge/OpenCorsairLighting)

[image]
[image]
[image]

Legion2, 2019-11-19:
I need the USB PID of SP100 and if it uses the same protocol as the Lighting Node PRO it should work.

Legion2, 2019-11-19:
I was able to find the ids of the LS100 see [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Corsair-fans-or-other-hardware#rgb-controller). Now i will try to get the LS100 working.

Comango104, 2019-11-20:
> I made this hardware and have used it in my pc for about a month now, the total cost per unit was around 7-10% (bought 10 at once since that's the lowest that makes sense), with both PCB and [...]

Wow that's awesome, I'll definitely send some friends to your github page as well.

Comango104, 2019-11-20:
> I was able to find the ids of the LS100 see [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-Corsair-fans-or-other-hardware#rgb-controller). Now i will try to get the LS100 working.

Wow that was fast! Thanks Legion. <3 I've been looking for a ambient lighting solution for a while now and for the price of some ws2812 strips, a powersupply and a ATmega32U4 MCU with proper control software this is perfect.

Legion2, 2019-11-23:
@Comango104 I also want to create an ambient lighting project. I created #68 so we can discuss ideas

PR #67 (2019-11-29): added support for LS100
created example to show how to create a corsair LS100 close #65
commit 0456348d (2019-11-29): added support for LS100 close #65
closed 2019-11-29 by Legion2