Legion2/CorsairLightingProtocol #73: Issue with compiling

KBhamm, 2019-11-27:
I succesfully flashed hoodloader.
but if i want to compile the code i get this error:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x20 pc=0x786d1e]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.(*SetupBuildProperties).Run(0xdebc68, 0x11cd65a0, 0xdebc68, 0x11f5c7f0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/setup_build_properties.go:93 +0xd2e
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0xdebc68, 0x11cd65a0, 0xdebc68, 0xdda500)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_setup.go:92 +0x54f
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x11cd65a0, 0x11c71f0c, 0x2, 0x2, 0xffffff01, 0x0, 0x4)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0x11c71f40, 0x11cd65a0, 0xdebbb4, 0x11c06ca8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:198 +0xbc
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:231
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:388 +0x772
C:\Program Files (x86)\Arduino\arduino-builder gab 2 zurück

Fehler beim Kompilieren für das Board CLP HoodLoader2 16u2.
```

Legion2, 2019-11-28:
Did you use Arduino IDE or the arduino-builder command? If you used the arduino-builder which command did you used?

KBhamm, 2019-11-28:
I used the newest IDE

Legion2, 2019-11-28:
Arduino IDE 1.8.10, Windows Store version or Arduino IDE 1.9.0 beta?

KBhamm, 2019-11-28:
IDE 1.8.10 from the arduino website.

KBhamm, 2019-11-28:
I can compile every code or load an empty code to the arduino. But If I want to compile the node pro sketch for the clp hoodload I get this error.

Legion2, 2019-11-28:
do you have installed the [HoodLoader2 Boards](https://github.com/NicoHood/HoodLoader2/wiki/Software-Installation#2-installing-board-definitions)?

KBhamm, 2019-11-28:
Yes, but I can make a clean install later after work and try again. Does it work on your end?

Legion2, 2019-11-28:
yes it works for me with my Arduino Mega

KBhamm, 2019-11-28:
Hmm well that's strange, I pick the arduino, the node pro the com port and the hoodloader clp and then it should just work with the node sketch. Or did I forget something?

Legion2, 2019-11-28:
Is there more output from the error?

KBhamm, 2019-11-28:
No only this. This was the whole error text.

Legion2, 2019-11-28:
which sketch exactly do you try to compile, did you modify the code?

KBhamm, 2019-11-28:
I tried the plain sketch. Just changed the output pins and the leds to ws2812, but I also tried the unmodified sketch, same result

Legion2, 2019-11-28:
What is the plain sketch? `LightingNodePRO`, `HoodLoader2UnoMegaController` or `HoodLoader2CLPBridge`? sketch is not the same as the device!

KBhamm, 2019-11-28:
i tried to upload this. sorry im really new to arduino and just want to make a second cheap node pro.
Board: CLP Hoodloader2 16u2
Device: Lighting node Pro
Port Com7: Arduino/genuino uno
`/*
   Copyright 2019 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingNodePRO.h>
#include <FastLED.h>
#define NUM_LEDS 50
#define DATA_PIN 8
#define DATA_PIN_CHANNEL_1 8
#define DATA_PIN_CHANNEL_2 9

CorsairLightingNodePRO cLNP;

void setup() {
	CLP::disableBuildInLEDs();
	auto ledController = cLNP.getFastLEDController();
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1>(ledController->getLEDs(0), ledController->getLEDCount(0));
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2>(ledController->getLEDs(1), ledController->getLEDCount(1));
}

void loop() {
	cLNP.update();
}`

Legion2, 2019-11-28:
this sketch should not be compiled for the Board CLP Hoodloader2 16u2. See [the guide for Arduino Uno/Mega](https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-use-on-Arduino-Uno-and-Arduino-Mega#upload-the-sketches) there it is explained which sketch you should use for Arduino Uno/Mega

KBhamm, 2019-11-28:
I managed it to work, had fun, learned something about programming and now i run really everything via corsair! :P 
How can i donate you something ? :)

KBhamm, 2019-11-28:
And I have one question left. Is it possible to turn off the leds after computer is shutting down? It just stops the effect and stays on.

Legion2, 2019-11-28:
You can setup hardware lighting in iCUE, which is used when the PC is off. Set the hardware Lighting to static color black.
[image]

For donations press the "Sponsor" button at the top of this page, there are different options you can choose from.
[image]

Legion2, 2019-11-28:
One hint, the Arduino Uno don't have that much of memory, so it could be, that you can not use as much led on both channels as shown in the plain `HoodLoader2UnoMegaController` sketch. I only tested the sketch on a Arduino Mega with more memory.

KBhamm, 2019-11-28:
Oh thanks didn't notice that this has a hardware profile, yeah. I think the memory is enough I only want to control 105 leds. Thanks for the hint.

Now I bought an official ls100 and it is crap. It freaks out my freesync on my monitor. Is it the same with the arduino one?

Legion2, 2019-11-28:
> Is it the same with the arduino one?
Do you mean Arduino Uno as LS100?

I'm currently building my own Ambient Lighting with LS100 using an Arduino clone and I have no problems with G-Sync.

KBhamm, 2019-11-29:
Yes or any other supported arduino.
The original Ls100 is totally not worth it 😅 I play on freesync and on stable 142 fps my freesync refresh rate is switching between any number from 44 to 144hz. It's totally out of controll.

KBhamm, 2019-11-29:
> You can setup hardware lighting in iCUE, which is used when the PC is off. Set the hardware Lighting to static color black.
> [image]
> 
> For donations press the "Sponsor" button at the top of this page, there are different options you can choose from.
> [image]

That doesnt work. it just stops the current animation and stays at this on full brightness :o

Legion2, 2019-11-29:
Sounds like a problem from iCUE not activating the HW Playback when shutting down.

Legion2, 2019-12-01:
What did you configure in the Hardware Lighting tab for the channel?

KBhamm, 2019-12-01:
Oh did I need to configure it while compiling? Then I overlooked it. I gonna do it again :P

Legion2, 2019-12-01:
no it's a setting in iCUE

Legion2, 2019-12-01:
How many leds did you set in you sketch and how many leds are configured in iCUE.
If you configured more leds in iCUE then you have in you sketch, HW does not work.

KBhamm, 2019-12-01:
No I have more leds in the sketch, 105 in sketch and maximum of 60 in icue per channel.

Legion2, 2019-12-01:
which version of the library do you use?

KBhamm, 2019-12-01:
If you mean with library your version then the last one 0.7.0, still my first arduino thing I've done :)

Legion2, 2019-12-01:
can you post your sketch here, so I can have a look.

KBhamm, 2019-12-01:
Thanks :) gonna do that tomorrow.

KBhamm, 2019-12-02:
> can you post your sketch here, so I can have a look.

i cant i didnt save it. but i do it the next time with an leonardo it should be easier and better then.

Legion2, 2019-12-05:
I will close this as the original Issue has been resolved. If you have further question or problem open a new one.

closed 2019-12-05 by Legion2