Legion2/CorsairLightingProtocol #74: Issue with Strip Scaling

cheeragpatel, 2019-11-29:
Hi, I have a single strip of 210 WS2182B connected and when I use the scaling function, the strip works for a few seconds and then starts blinking all on/off. When I set the scale back down to 150 it seems to start working again. Any idea?

Code I'm using:
```
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2

CRGB ledsChannel1[210];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  FastLED.addLeds<NEOPIXEL, DATA_PIN_CHANNEL_1>(ledsChannel1, 210);
  ledController.addLEDs(0, ledsChannel1, 60);
  ledController.onUpdateHook(0, []() {
    CLP::scale(&ledController, 0, 210);
  });
}

void loop() {
  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}
```

Legion2, 2019-11-29:
Maybe a problem with your power supply? How do you power the strip and which Arduino do you use?

cheeragpatel, 2019-11-29:
I'm powering the strip via an external power supply and I'm using a Leonardo. I haven't had issues driving the strip with this setup before as I was using this exact setup for a different lighting setup and it was working fine. I did some more testing and the issue happens more with rainbow mode. I think the arduino might be getting overloaded and dropping data.

Legion2, 2019-11-29:
you only define 60 leds in iCUE so it should not be overloaded with this.

Legion2, 2019-12-02:
Which version of this library (CorsairLightingProtocol) do you use?

cheeragpatel, 2019-12-04:
Figured it out, my power supply was going bad, tried with another one and now its functioning correctly.

closed 2019-12-04 by cheeragpatel