Legion2/CorsairLightingProtocol #75: Fan noise while using PWM on Commander PRO

Gannit, 2019-11-30:
I have built a circuit for the commander PRO and I am using the PWM signal to drive an N channel MOSFET to run the 12v fans pulsing the negatice rail. I use the same technique on my 3D printers.  However anything less then a 100% duty cycle spins the fans ok but makes the fans noisy where you can here the PWM pulsing from the fan. like a background buzzing. I have tried multiple types of fans. 

I Tried changing the fan update rate from 500 to a lower value but it didnt seem to make any difference.

Legion2, 2019-11-30:
which pins do you use for the fan pwm signal?

Legion2, 2019-11-30:
Also which board do you use? The PWM timings depend on the boards used, I only tested it with Pro Micro.

Gannit, 2019-11-30:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/Legion2/CorsairLightingProtocol/issues/75?email_source=notifications\u0026email_token=AN5CK4SGMIITUR4IQ4PFVP3QWJUSZA5CNFSM4JTGTG32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFQHHYI#issuecomment-559969249",
> "url": "https://github.com/Legion2/CorsairLightingProtocol/issues/75?email_source=notifications\u0026email_token=AN5CK4SGMIITUR4IQ4PFVP3QWJUSZA5CNFSM4JTGTG32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFQHHYI#issuecomment-559969249",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

I connected the fans to pins 5, 6 and 9 guided by the code in the examples Sent from my Samsung Galaxy smartphone.

Gannit, 2019-11-30:
Ah I am using a pro micro however not the spark fun version. It could be that. When programming I have to set it as a leonardo as when I choose pro micro, it wont upload. The ones I have here are by keeyees (a Chinese brand I believe)I will get hold of a spark fun board and try with that. Thank you  Sent from my Samsung Galaxy smartphone.

[
{

Legion2, 2019-12-16:
@Gannit Did you try with the other board?

Gannit, 2019-12-16:
I managed to get hold of a full size leonardo and it worked ok.  So I'm guessing it could be the cheap clone arduino causing the issue. Sent from my Samsung Galaxy smartphone.

[
{

Legion2, 2019-12-16:
Ok i will then close this Issue, if you have any other problem or suggestion, you can open a new one.

closed 2019-12-16 by Legion2