Legion2/CorsairLightingProtocol #80: Add support for WS2801?

BigDi, 2019-12-05:
Ì have some meters of WS2801 at home. Is it possible to add support for this kind of LED stripe?

Legion2, 2019-12-05:
They are supported see https://github.com/FastLED/FastLED/wiki/Overview#supported-chipsets. For more details on how to use them look in the documentation of [FastLED](https://github.com/FastLED/FastLED).

closed 2019-12-05 by Legion2

BigDi, 2019-12-05:
I know that they supported with FastLED but how to configure in CLP?

Legion2, 2019-12-19:
@BigDi Sry I missed your reply. I does not own WS2801 LEDs and don't know how to set them up. But your example looks good to me. What beside the fake that 32U4 has only one SPI interface is the problem?

koko009, 2019-12-20:
i would like you for this work 
but i have issues 
1- ihave fan 16 led when apply ll series works but There is a difference in a number 4led 
2-when apply ls100 after upload sketch arduino Leonard is not work and sometimes working but when selected single display from hardware arduino disconnected and not working again i have 2 arduino Leonard one uploaded node pro and the other ls100

Legion2, 2019-12-20:
@koko009 
1. see https://github.com/Legion2/CorsairLightingProtocol/wiki/How-to-transform-LL-Fans-format-to-LED-strip how to fix the led order issue 
2. which sketch did you use?

koko009, 2019-12-20:
sketch ls100 to arduino Leonard

Legion2, 2019-12-21:
What do you mean with not working? Is the device not detected by iCUE or does the LEDs not light up? Which version of the CorsairLightingProtocol library do you use? And did you modify the uploaded code?

koko009, 2019-12-21:
no i dont modify code 
not working arduino not detected by icue and not detected by windows but when upload code node pro then replug usb and replug again and replug again to finish upload code node pro then working fine 
version 0.8 
but i want to tell you 
i have 2 arduino Leonard  
one use code node pro and working fine 
tow use code ls100 and not working 
i want to use multiple devices 
and i want to use ambilight via icue 
thank you

Legion2, 2019-12-22:
For using multiple devices see https://github.com/Legion2/CorsairLightingProtocol/blob/dev/README.md#use-multiple-devices