Legion2/CorsairLightingProtocol #84: add off button for LS100

theorgBigBoss, 2019-12-17:
Hi
First of all thanks for your great job ^_^
i use LS100 in ProMicro board , when i turn off PC ls100 still work
i realize the orginal ls100 have a poweroff button on it for this situation
its possible to add this button to LS100?
(sorry for my bad english >_<)

Legion2, 2019-12-17:
There are three options to solve this:
1. Set a hardware Playback profile in ICUE to static black
2. Add a bool variable to store the ON state of the Arduino, add a button that change this state and turn off all leds. Only if the state is on call the `FastLED.show()` method.
3. Add a button, when pressed set the ChannelMode of all channels to Disabled using the LEDController.

Legion2, 2019-12-18:
@theorgBigBoss which option do you prefer?
As I just saw option 3 is not possible, because the method `LEDController.setLEDMode()` is not public.

theorgBigBoss, 2019-12-22:
> 
> 
> @theorgBigBoss which option do you prefer?
> As I just saw option 3 is not possible, because the method `LEDController.setLEDMode()` is not public.

i think add a bool variable will be fine, FastLED.show()...

PR #86 (2019-12-23): Added ON/OFF button to LS100
Added an optional hardware button to disable the lighting of the LS100 independent of iCUE close #84.
The button should be connected to pin 4 of the Arduino and connect pin 4 to GND if pressed.
closed 2019-12-23 by Legion2