Legion2/CorsairLightingProtocol #90: Issue with LS100 and clone pro micro

bryan065, 2019-12-30:
**Describe the bug**
Using a clone Pro Micro (5v), when uploading the LS100 example sketch, the lighting channels don't work or light up (sometimes will work with rainbow swirl effect but cannot be changed in iCue). The Pro Micro is detected in iCue as an LS100 device.

When uploading any other sketch (Node Pro, Node Core, Commander Pro), everything works as expected and lighting channels can be controlled in iCue.

When uploading a Node Pro sketch but choosing "Smart Lighting Controller" as the device, the lighting channels can be controlled and the Pro Micro is detected as a LS100 device but the gamma correction from release 0.9.0 isn't applied.

I'm using a clone pro micro and reflashed the bootloader so I don't have to short the RST pins to upload any sketches.

Doesn't matter which pins were used, although I've had the sketch work once with the exact same configuration but with release 0.8.0 and board library 0.1.1. Suddenly stopped working again

**To Reproduce**
Steps to reproduce the behavior:
1) Choose "CLP SparkFun Pro Micro 5V, 16MHz"
2) Choose "Smart Lighting Controller"
3) Upload example "LS100" sketch

**Images**
[image]
Current wiring and setup. Shouldn't be an issue as using the Node Pro sketch is working as expected

**System (please complete the following information):**
 - OS: Windows 10 x64 1909 & Windows 10x64 1903
 - Board: Clone Pro Micro 5V, 16MHz
 - IDE: Arduino IDE 1.8.10
 - Version: 0.8.0 & 0.9.0, Board Library 0.1.1 & 0.1.2
 - Sketch: LS100

**Code changes**
Tested sketch with default pins but same issue. Preferred pins are 9 & 10 for the lighting channels and pin 14 for the LS100 button.

Legion2, 2019-12-30:
Did you try to disable the on/off button feature?

bryan065, 2019-12-30:
Yup, I've tried version 0.8 before the on/off button and version 0.9 with the on/off button. I've removed the on/off code from 0.9 but it didn't make a difference.

The on/off button is working as expected because when channels do light up, the button can turn it on/off (but I can't control the lights in iCue as mentioned)

As soon as I upload the node core sketch, everything works. If I add the on/off button code to the node core sketch, everything still works.

Looking at the code, I think there's a difference between `corsairLightingNodeCOREFirmware()` and `corsairLS100Firmware()` that's causing it but I haven't been able to find it yet. I'm still studying how the protocol works.

Legion2, 2019-12-30:
The different firmware names are only aliases. So there is no difference in semantics.

Legion2, 2019-12-30:
Did you apply the gamma correction in your sketch?

bryan065, 2019-12-30:
Nope, I haven't touched anything else in the sketches. I'm still learning how it all works. I was just planning on seeing the differences between the node pro sketch and ls100 sketch to figure out why the ls100 one doesn't work. 

Worst case is that I just use the node core sketch instead and apply the gamma correction to that

VFjr, 2019-12-31:
I can confirm that I have the exact same issue. Tested on both 8Mhz 3v3 Pro Micro, and a custom 16Mhz 5V 32u4 board, both of which work using the Lighting Node Pro sketch. Using the Arduino IDE 1.8.10, and also tested 0.8.0 & 0.9.0, and Board Library 0.1.1 & 0.1.2. I did not make any changes to the code, still using the default pins.

The Lighting channels don't work, and I do seem to get sometimes the rainbow swirl effect that is not controlled by iCue. The button seems to sometimes turn the leds off when there is a rainbow swirl, but not always.

Legion2, 2019-12-31:
Ok seems like the LS100 sketch uses too much memory. The total size of both led channels must be reduced when using an ATmega32U4. This can also be seen in the real LS100 there you can't have two channels with 135 leds, you can only have one channel with 135 and the second only with 54 leds. I will update the LS100 sketch and add a hint.

closed 2019-12-31 by Legion2

theorgBigBoss, 2020-02-24:
> [image]
Hi
this is 3D sketch for print?

Legion2, 2020-02-24:
@bryan065 can you share the STL of the LS100 enclosure?

AlpineWhite, 2020-03-14:
In lieu of OP's design, I am developing a snap-closed enclosure.  Once I figure out how I want to route power in, I will post the STLs here.
[image]
[image]
[image]

Heavily inspired by OP, with a few tweaks and a cover.

AlpineWhite, 2020-03-15:
Grabcad Link:
https://grabcad.com/library/ls100-pro-micro-omron-3-5mm-1
Model is built around using JST XH connectors, Omron 6x6mm microswitch, and Wurth Electronik 
694102303002 which is a 3.5mm power input.  3.5mm jack size allows selection of 12V (WS2815) or 5V (WS2813/2812B) LED strips.  The board and connector assembly slides in from the 'top' and then the cover locks the board in place.  Recommended superglue or liberal amounts of hot glue to secure the power input.  LS100 button is programmed as pin 18, and leg closest to USB is bent out and connected to GND.  Other two legs are meant to be clipped.  Tolerances are tight, so some post-processing may be required depending on your printer.

PR #91 (2019-12-31): fixed out of memory bug in LS100
reduced the leds on the second channel(first channel in iCUE) close #90
formatted the code of the LS100 sketch
commit 178429ee (2019-12-31): fixed out of memory bug in LS100 close #90