Legion2/CorsairLightingProtocol #93: STM8S103F3 microcontroller

theorgBigBoss, 2020-01-01:
Hi
I found this STM8S103F3 microcontroller , it support V-USB (description here : https://hackaday.com/2014/03/02/software-usb-on-the-stm8/)
Here is specification of STM8S103F3 microcontroller : https://www.st.com/en/microcontrollers-microprocessors/stm8s103f3.html

can we use this cheap micro STM8S103F3 instead of Atmega32u4?

Legion2, 2020-01-01:
To use STM8S103F3 you need the 3rd party Board definition from https://github.com/tenbaht/sduino but this has some limitations as the author describes:
> I adopted the Arduino core functionality for the STM8S to set up a simple programming environment. But unfortunatly there is no free C++ compiler for these CPUs. This makes it impossible to do a full port of the whole enviroment and integrate it 100% in the Arduino world as it has been done for the STM32 and the ESP8266.

So the libraries must be written in C, but this library(Corsair Lighting Protocol) and FastLED are written in C++. V-USB demonstrate that it is possible to do USB, but it does not implement Arduino USB API, so the complete HID stack must be reimplemented.

It could be possible to use STM8S103F3 but it is not a drop-in replacement for the Atmega32u4 and a lot of work has to be done. I will not support STM8S103F3 in Corsair Lighting Protocol, because it's completely out of scope.

closed 2020-01-01 by Legion2

TylerSeiford, 2020-01-01:
On a tangent topic, how hard would it be to support alternative arduino-compatible microcontrollers?

I know you can use some microcontrollers like the ESP8266 and STM32 in the arduino IDE, but would it be possible to create a CLP program running on one?

Legion2, 2020-01-01:
You can try to compile the CLP for such a board and if it compiles it's like that it will work. For this "test" you don't even need the hardware.

Legion2, 2020-01-25:
The RawHID part is only compatible with AVR and must be implemented for other platforms as teensy3/4. The ESP8266 don't support USB HID, but he ESP32 does support HID via Bluetooth and I implemented an adapter for it. But iCUE does not use the Bluetooth device.