Legion2/CorsairLightingProtocol #95: Add support for customizable gamma for LS100

bryan065, 2020-01-01:
Since #90 fixed issues with OOM with the atmega32u4, next enchancement I see is to add the ability to change the gamma correction but have it default to 2.0 (currently it's hardcoded to 2.0) and only have it affect the video lighting mode.

With the specific strips I currently have, some only need a correction of 2.2 but some need as high as 3.0 or 3.5 (when being used with adalight instead of CLP).

Legion2, 2020-01-01:
You can simply apply custom gamma correction or any other post processing directly in any sketch by using the `onUpdateHook`, there you have access to the led array and can manipulate the colors before they are displayed by FastLED. The gamma correction is currently hardcoded, because it's very fast to compute this way and better than nothing. Finding the correct gamma value for a strip is very hard and even with the correct gamma value, the colors on the strips don't look "real". So as stated in the documentation of the gamma correction function, it's only an approximation.

Btw. the gamma correction applies to every lighting on that channel, not only the "video lighting mode". The problem is, iCUE does not set the leds to max brightness even if you set the color #FFFFFF and 100% brightness in iCUE. You only get #7F7F7F on the led strip and if you then apply gamma of 2.0 you will get only #3F3F3F.

@bryan065 If you want your custom color correction and you know how you can transform the color values, you can do that in the `onUpdateHook`. You can use the `applyGamma_video` function of FastLED to apply the gamma on the complete LEDs array.

Legion2, 2020-01-01:
By default no gamma correction is applied in the LS100 sketch because of the fact, that it would decrease the brightness. iCUE do some preprocessing and does not scale the brightness linear. The device brightness setting in iCUE does the following:

| iCUE brightness Setting | Brightness send to Arduino | Brightness after gamma correction (2.0) |
| ------------- | ------------- | ------------- |
| 100% |  50% | 25% |
| 67%  | 25%  | 6.25% |
| 33% | 6.25% | 0.39% |
| 0% | 0%  | 0% |

Note the device brightness setting in iCUE does not apply to HW Lighting.

Legion2, 2020-01-10:
@bryan065 did you try to apply the gamma correction with the `onUpdateHook` and `applyGamma_video`?

commit 974fb858 (2020-01-18): added example for customizable gamma close #95
closed 2020-01-18 by Legion2