Legion2/CorsairLightingProtocol #97: teensy 3.2 - 3.6 support?

Muncky, 2020-01-04:
Hi,
i was wondering if the teensy boards would work. i think it should be possible since those chips have USB on board.
[https://www.pjrc.com/teensy/schematic.html]

Legion2, 2020-01-04:
I have no teensy board, so it's difficult to debug the code. But i will look into it.

Legion2, 2020-01-04:
Teensy does not support Arduino IDE correctly. They require a plugin (custom version of Arduino IDE). And I can't find the boards definitions for the teensy boards, so I can't even try to compile the code for the teensy.

Muncky, 2020-01-05:
it only requires a plugin to program the chip. the board has it's own bootloader and most arduino libraries work for teensy as well. i have a few of them, in fact i prefer teensy over arduino.

Legion2, 2020-01-05:
But to program non Arduino chips it should not be require a plugin. Arduino IDE has an own system to handle the different boards, by simply adding a configuration file. I opened PaulStoffregen/cores#412 to improve the compatibility of teensy with the default Arduino IDE.

FrankBoesing, 2020-01-05:
Without even knowing what a Teensy is... take your time. Buy one and be amazed how much better it is :)

Muncky, 2020-01-05:
I'll see if I can figure something out... but I must warn you... I'm only a hobbyist in electronics and programing... ;-) 
I'll let you know if i get somewhere...

Legion2, 2020-01-05:
@FrankBoesing better hardware or better software (api/libraries)?

Legion2, 2020-01-17:
I installed Teensyduino and tried to get it working with CLP. I discovered multiple problems:
1. The USB VID and PID can't be set using the Boards definition, because all ids are defined in [usb_desc.h](https://github.com/PaulStoffregen/cores/blob/master/teensy3/usb_desc.h)
2. The Raw hid interfaces are very basic see [usb_rawhid.h](https://github.com/PaulStoffregen/cores/blob/master/teensy3/usb_rawhid.h)
3. Naming conflict with `RawHID` global variable
4. The USB stack is not modular as the USB Core of Arduino
5. Because of 1. and PaulStoffregen/cores#412 I can't add a board definition to CLP Boards for teensy
6. The integration in the Arduino Boards Manager seems like a hack from teensy (can be done better/correct)

Legion2, 2020-03-03:
Until there is an Arduino Boards Definition for Teensy Boards released I can't work on the support for teensy.

Legion2, 2020-09-30:
I started a thread at Teensy forum (https://forum.pjrc.com/threads/63388-Support-Arduino-Core-API-and-Board-definitions) to ask them about the better integration of the Arduino ecosystem. I explained the benefits for the community and proposed they should support the [Arduino Core API](https://github.com/arduino/ArduinoCore-API) and [Board definitions](https://arduino.github.io/arduino-cli/latest/platform-specification/) for Teensy boards. But they silently deleted my thread and when I requested the reason for that they banned me for spamming. (I only created one thread and then ask an admin about it and therefore I have been banned for spamming!!!)
[image]

So my conclusion is that the people behind Teensy don't want to help the community and just want to have their own isolated ecosystem. They don't want that their users benefit from the arduino open source ecosystem with many useful libraries and tools. They don't even want that their users know about it.

Muncky, 2020-10-02:
Hey Leon,   that&#39;s not nice of them... i guess they really want to protect there product... not really open scource then.... also there bootloader on the chip is nowhere near open scource... a real secrect..  however there is a rawHID option for teensy in arduino IDE that sends and receives messages... maybe i can get it to work with that... but then i&#39;ll need to figure out your library completly. that can take alot of time because i&#39;m not all the time busy with coding. as i understand, you could get any microcontroller with native usb function to mimic the icue products as long as you send the Icue program the correct information back.   i actually bought the arduino micro with the  ATmega32u4 , i&#39;ll have a go with that ;)   thanks for the effort anyway...    Kind Regards,  Met vriendelijke groeten,   Pieter Demuynck     Dnia 30 września 2020 09:49 Leon Kiefer &lt;notifications@github.com&gt; napisał(a):   I started a thread at Teensy forum ( forum.pjrc.com forum.pjrc.com ) to ask them about the better integration of the Arduino ecosystem. I explained the benefits for the community and proposed they should support the  github.com Arduino Core API  and  arduino.github.io Board definitions  for Teensy boards. But they silently deleted my thread and when I requested the reason for that they banned me for spamming. (I only created one thread and then ask an admin about it and therefore I have been banned for spamming!!!)  user-images.githubusercontent.com  So my conclusion is that the people behind Teensy don&#39;t want to help the community and just want to have their own isolated ecosystem. They don&#39;t want that their users benefit from the arduino open source ecosystem with many useful libraries and tools. They don&#39;t even want that their users know about it.

Legion2, 2020-10-02:
@Muncky the problem about teensy USB HID interface is that it is not open such as the one of arduino. There are only a handful of predefined profiles which you can select in the IDE. You would have to change the teensy code to modify the USB Reports and use other USB IDs. See this comment https://github.com/Legion2/CorsairLightingProtocol/issues/97#issuecomment-575800708.

Legion2, 2022-09-03:
In a recent teensy release (https://www.pjrc.com/teensyduino-1-57-released/) support for Arduino Boards Definition was added.

closed 2022-09-26 by github-actions[bot]