Legion2/CorsairLightingProtocol #105: Fan support for Commander Pro

Eriobis, 2020-01-25:
**Describe the problem**
A clear and concise description of your problem.

I am setting the commander pro but there's no control for the fans, is this feature implemented yet ?

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System (please complete the following information):**
 - OS: Windows 10 
 - Board: Sparkfun Micro 5V
 - IDE: Arduino IDE
 - Version: 1.8.10
 - Sketch: CommanderPRO]

**Code changes**
Did you changed the code of the library or the sketch?
Nop

Eriobis, 2020-01-25:
I have found how ... might be nice to have it in the wiki though !! I tought the feature wasn't implemented yet

closed 2020-01-25 by Eriobis

Legion2, 2020-01-25:
How did you solved the problem? If you are using the CommanderPRO sketch there should be 4 fans in iCUE.

Legion2, 2020-01-25:
The Commander PRO has some limitations, I will document them in the sketch and Wiki #106

Legion2, 2020-03-05:
@Eriobis I added some documentation in the [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/Commander-PRO)