Legion2/CorsairLightingProtocol #107: Wiring

eduardokum, 2020-01-25:
there is some documentation on how to connect the LED wires to control 6 fans on a lighting node pro channel and control them individually

I use a hub but it distributes only the changes of the first fan of the icue, as if all the fans were cloned from the first.

is there any way to connect them so that I can control them individually

I use sketch LightingNodePro on an arduino pro mini board with leonardo microcontroller

Legion2, 2020-01-26:
What hub did you use? If you created the hub yourself, the fans should be in series, as you would connect multiple led strips in series. That's because the RGB control connector has four cables, 5V, In, Out, GND (In/Out may be swapped!). The Out singal should be connected to the In of the next fan.

eduardokum, 2020-01-26:
i'm using this fan, Deepcool CF120

[image]

with this hub, the come with the box, this one in the center
[image]

Yes, I understand, but my deepcool cf120 fans only have 3 pins, 5v Data gnd, the LL fans from corsair also have an output signal pin that allows serial connection is that it?

Legion2, 2020-01-26:
yes all Corsair fans have 4 pin RGB connectors

Legion2, 2020-01-26:
I think your fan hub does not support to address individual fans.

eduardokum, 2020-01-26:
Yes! I don’t know that! 

Thanks!

closed 2020-01-26 by eduardokum

skadzynski, 2020-01-27:
[image]
 Hi
Meine Lüfter haben die Kontakte, was ist denn was?

Legion2, 2020-01-27:
Sind das addressierbare LEDs? Welche Spannung brauchen die LEDs? Wenn es addressierbare LEDs sind und die Spannung 5v ist würde ich tippen, das außen die 5V und GND Anschlüsse sind. Und die Anschlüsse dazwischen sind in und out signal. Du musst das ausprobiere es sehr wahrscheinlich ausprobieren wenn du keine anderen Informationen über die Lüfter hast.

skadzynski, 2020-01-27:
Ja sind Adresierbare, +,-,data in is klar aber das 4. Was unbeschriftet ist is un klar, wenn ich es als data out nehme wird der 2. Lüfter nicht gesteuert.

skadzynski, 2020-01-27:
[image]
 Das sind die, ich googel schon seid 2 Tagen 😉
Dein Projekt ist echt der hammer👍

Legion2, 2020-01-27:
@skadzynski waren bei dem Lüfter irgendwelche Kabel und Adapter dabei? Vielleicht geben diese Aufschluss darauf was der vierte Anschluss macht.

skadzynski, 2020-01-28:
[image]
Nur der hub, fernbedienung und 6 pin Lüfter 
[image]

robbo247, 2020-01-28:
hope this helps beenworkin on different 6 pin hubs the past few days,these
are the pinouts BUT if your making your own cables beware of the 12v

Legion2, 2020-01-28:
@skadzynski did you use the 6pin connector to test the your fans? 

@robbo247 What is the pin out? Seems your message was broken.

robbo247, 2020-01-28:
[image]

robbo247, 2020-01-28:
[image]

robbo247, 2020-01-28:
[image]
[image]

skadzynski, 2020-01-28:
Ich werde meinen Controller auch mal zerlegen wenn ich Zeit finde, und dann berichten.