Legion2/CorsairLightingProtocol #108: HoodLoader2UnoMegaController

skadzynski, 2020-01-28:
**Describe the problem**
A clear and concise description of your problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino Leonardo]
 - IDE: [e.g. Arduino IDE]
 - Version: [e.g. 0.3.0]
 - Sketch: [e.g. SingleStripLightingNodePRO]

**Code changes**
Did you changed the code of the library or the sketch?

skadzynski, 2020-01-28:
Hi
Ich hab da ein Problem, bin noch extrem Anfänger.
Hab mit ls100 auf einem arduino UNO und die RGB ist vertauscht, kann ich das ändern?

Legion2, 2020-01-28:
Du meinst die Farben sind vertauscht? Dann musst du statt `FastLED.addLeds<NEOPIXEL, DATA_PIN_CHANNEL_1>(ledsChannel1, CHANNEL_LED_COUNT);` die `addLeds` funktion benutzen, die drei Argument hat. Also `FastLED.addLeds<WS2812, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, CHANNEL_LED_COUNT);` wobei `GRB` die Reihenfolge der Farben ist.

skadzynski, 2020-01-28:
Ok werde ich morgen direkt machen wenn ich von der Arbeit komm, super vielen dank für deine Hilfe, also grad die Zeilen austauschen.👍

skadzynski, 2020-01-30:
Hi
Vielen Dank für deine Hilfe, hat natürlich funktioniert.👍

closed 2020-01-30 by skadzynski