Legion2/CorsairLightingProtocol #109: ESP controller support?

MajorLOLGaming, 2020-01-30:
Support for ESP devices like [this one](https://www.ebay.com/itm/NodeMcu-Lua-WIFI-Internet-Things-Development-Board-Based-ESP8266-CP2102-Module-S/191736193319?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) would be a great thing. If you need hardware for testing, i can test some for you.

MajorLOLGaming, 2020-01-30:
The NodeMCU 1.0 in particular.

Legion2, 2020-01-30:
The ESP8266 have an external USB chip and only supports Serial communication. But this library requires more control over the usb connection like creating new USB Endpoints, USB Reports and USB HID support. This is a hardware of the ESP8266 does not support this, so it can't work with Corsair Lighting Protocol.

closed 2020-01-30 by Legion2

MajorLOLGaming, 2020-01-30:
Guess I need this? https://www.ebay.com/itm/184025796954

Legion2, 2020-01-30:
yes

MajorLOLGaming, 2020-02-07:
It is ordered. I look forward to using your project :)

MajorLOLGaming, 2020-03-12:
It arrived roughly a week ago, and it all works great. Thank you.

Legion2, 2020-03-12:
You are welcome

Wolvenmoon, 2021-07-23:
I had a thought that I want to jot down regarding this. I don't expect you to do this yourself, but I wanted to write it down where other people wondering about doing wifi-connected CLP-compatible controllers could see just in case someone else wanted to.

I think that requiring a USB connection to the host does away with a major strength of the ESP chips, which is their wireless capabilities.

So, here's what I'm thinking.

On the ESP8266/ESP32, an instance of CLP running as a server waiting for commands via wi-fi (or Bluetooth, maybe).

On the host running iCUE, likely Windows, an application that can generate any number of virtual CLP-compatible devices with user-definable IDs and translate commands from Corsair iCUE into commands sent over wi-fi (or Bluetooth) to the ESP nodes running CLP.

I'm completely new to messing with this kind of programming, so I'm not sure I'm the person to attempt it, but if it's something you feel would be useful, I'll try to take a swing at it as time permits.

Legion2, 2021-07-23:
I already though about this and also started digging around writing a custom windows driver which can emulate CLP devices and act as a server. However, writing a windows driver is complex and I have no code jet.
If you get something working it would be very helpful. Here is the guide I was trying to use: https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/writing-a-ude-client-driver

sle118, 2022-04-20:
Just my two cents here.

The esp32-s2 provides the usb driver capabilities now