Legion2/CorsairLightingProtocol #110: Save effect

eduardokum, 2020-02-07:
**Describe the problem**
Hello, when I save effect on the hardware and close the icue it looks different, it seems that it doesn't persist

**Screenshots**
iCue Settings
[image]

Preview settings when iCue is open
https://s5.gifyu.com/images/certo.gif

After I close iCue
https://s5.gifyu.com/images/errado.gif

 - OS: Windows 10
 - Board: Arduino Leonardo
 - IDE: Arduino IDE
 - Version: 0.10.0
 - Sketch: LightingNodePRO

Legion2, 2020-02-07:
This is a problem of the Rainbow channel effect, it should react to the size of the led strip, but iCUE does not provide this information. I will test this with a real Commander Pro but I think it has the same problem. So nothing I can fix in general. You can use other rainbow effect, they should look more similar to the preview.

Legion2, 2020-02-07:
This also happens with a real ComPro, so it's a problem of Corsairs iCUE.

Legion2, 2020-02-07:
You can change the library file so you get the effect you want. Change the end of line in the file [src/FastLEDController.cpp#L118](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/src/FastLEDController.cpp#L118) from `-3 : 3` to `-15 : 15`. You can find the installed library in your Arduino Library folder.

eduardokum, 2020-02-07:
> You can change the library file so you get the effect you want. Change the end of line in the file [...]

did it work, does it affect the other effects?

Legion2, 2020-02-07:
@eduardokum this setting affects the Rainbow Channel and Rainbow Wave effect. 3 is the value of the Rainbow Channel and 15 for the Rainbow Wave effect.

Legion2, 2020-03-03:
This Issue currently can't be fixed, as there is no mean in the Corsairs protocol to determine the type of connected device. So some effects can't be rendered correctly for some devices, e.g. Fans of different sizes. This Issue can also be reproduced with real Corsair LNP.

closed 2020-03-03 by Legion2