Legion2/CorsairLightingProtocol #111: Mulitple devices on the same microcontroller

Comahar, 2020-02-17:
Is it possible to make a board that shows up as 2 devices in iCUE?
For example, CommanderPro and LightingNodeCore on the same board because this makes a good combo to control all lights of fans and their speeds.
This may not be possible on all boards because of the IO and memory limitations but it will be cool if we don't have to use 2 microcontrollers for this job.

McHauge, 2020-02-18:
could you not just program one as a CommanderPro and that can control it all in itself, why would you need two unless you want to control even more LEDs with it?

Comahar, 2020-02-18:
In my setup, my fans have data in and no out, so they can't pass the data. In this situation, the only solution I can think of is using different pins for every fan and extra led strips.

Is there a way we can split the led channel data?
Perhaps by changing this part of the CommanderPRO's code, we can split the led data to multiple pins, but I can't really see a way it is easily done.

```
CorsairLightingFirmware firmware = corsairCommanderPROFirmware();
ThermistorTemperatureController temperatureController;
FastLEDController ledController(&temperatureController, true);
SimpleFanController fanController(&temperatureController, FAN_UPDATE_RATE, EEPROM_ADDRESS + ledController.getEEPROMSize());
CorsairLightingProtocolController cLP(&ledController, &temperatureController, &fanController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

CRGB ledsChannel1[CHANNEL_LED_COUNT];
CRGB ledsChannel2[CHANNEL_LED_COUNT];

void setup(){
	FastLED.addLeds<NEOPIXEL, DATA_PIN_CHANNEL_1>(ledsChannel1, CHANNEL_LED_COUNT);
	FastLED.addLeds<NEOPIXEL, DATA_PIN_CHANNEL_2>(ledsChannel2, CHANNEL_LED_COUNT);
	ledController.addLEDs(0, ledsChannel1, CHANNEL_LED_COUNT);
	ledController.addLEDs(1, ledsChannel2, `CHANNEL_LED_COUNT);
}
```

Legion2, 2020-02-18:
A CommanderPro can control the fans speed and fans lights. Corsair RGB fans have a Data in and out, so you can chain fan lighting signal. This way you can connect 6 fans on one LED Channel.
But if your fans don't have a data out, you can split the signal in the arduino code. `ledsChannel1` contains all RGB data as array, you just need a pointer offset for the different fans RGB values and pass this pointer to FastLED, to create a new "virtual" led stip for a single fan.

Comahar, 2020-02-18:
So I did what you said and it works well. But I still think that it will be cool to use a single microcontroller and have more channels, so something like this setup would be possible.

[image]

For anyone who is wondering about the code, you can do it like this.
```
FastLED.addLeds<NEOPIXEL, LED_FAN_PIN_1>(ledsChannel1, 0, LEDS_PER_FAN);
FastLED.addLeds<NEOPIXEL, LED_FAN_PIN_2>(ledsChannel1, LEDS_PER_FAN, LEDS_PER_FAN);
FastLED.addLeds<NEOPIXEL, LED_FAN_PIN_3>(ledsChannel1, 2*LEDS_PER_FAN, LEDS_PER_FAN);
FastLED.addLeds<NEOPIXEL, LED_FAN_PIN_4>(ledsChannel1, 3*LEDS_PER_FAN, LEDS_PER_FAN);
```

Legion2, 2020-02-18:
This is not possible with a 32u4 based board, because the USB functionality is part of special hardware in the 32u4 chip and this only can be one USB device. Maybe there are other chips that can be detected as multiple USB devices, but I don't know any.

McHauge, 2020-02-18:
hope you know that the fan RGB hub is just a fancy cable in a box right, there is no controller in it, just daisy-chain the fans data-in and data-out from one fan to the next.
[image]

here is a bisic diagram of how that can be done: 
[image]

TylerSeiford, 2020-02-18:
> This is not possible with a 32u4 based board, because the USB functionality is part of special hardware in the 32u4 chip and this only can be one USB device. Maybe there are other chips that can be detected as multiple USB devices, but I don't know any.

Are you sure?  I've messed around with projects using a single 32U4 as multiple HID joysticks before.

Legion2, 2020-02-18:
I think RAW HID uses the USB Control Endpoint for out Reports. So if there are two LNPs for example on the same device, they get the same USB HID Reports, because they share the USB Control Endpoint.
I think this can be fixed if the RAW HID implementation uses Report IDs but currently it does not and I don't know if iCUE would send Report IDs.

closed 2020-04-19 by stale[bot]

t0mb3r, 2020-10-03:
Hello, I'm trying to create virtual strips to assign to different fans that doesn't have data out but I am having no luck on making it work. this is what I've changed from the LightingNodePRO code: 

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3
#define DATA_PIN_CHANNEL_3 3
#define DATA_PIN_CHANNEL_4 3
#define DATA_PIN_CHANNEL_5 3

#define CHANNEL_LED_COUNT 96
#define LEDS_PER_FAN 20

CRGB ledsChannel1[CHANNEL_LED_COUNT];
CRGB ledsChannel2[4*LEDS_PER_FAN];

...

void setup() {
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, CHANNEL_LED_COUNT);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 0, LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_3, GRB>(ledsChannel2, LEDS_PER_FAN, LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_4, GRB>(ledsChannel2, 2*LEDS_PER_FAN, LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_5, GRB>(ledsChannel2, 3*LEDS_PER_FAN, LEDS_PER_FAN);
  ledController.addLEDs(0, ledsChannel1, CHANNEL_LED_COUNT);
  ledController.addLEDs(1, ledsChannel2, 4*LEDS_PER_FAN);
}

Legion2, 2020-10-04:
@t0mb3r have a look at https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/MultipleFans/MultipleFans.ino