Legion2/CorsairLightingProtocol #114: problem with color

theorgBigBoss, 2020-03-07:
Hi
i use WS2812B strip
today i realise when i set instant color to white it shows Yellow and when i set set it to light blue led show white

**Screenshots**
https://imgur.com/RrdI2Et
https://imgur.com/tZ1rnll

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino Leonardo]
 - IDE: 1.8.10
 - Version: 0.10.0
 - Sketch: LNP

Legion2, 2020-03-07:
This is not a bug, it's because of the bad calibration of the LED strip. See [this page](https://github.com/FastLED/FastLED/wiki/FastLED-Color-Correction) and [API reference](http://fastled.io/docs/3.1/group___color_enums.html) of FastLED on how to correct the color. Use `setCorrection()` and `setTemperature()` in the `setup` function.

closed 2020-03-07 by Legion2

theorgBigBoss, 2020-03-08:
Many thanks dude
ill try it.

theorgBigBoss, 2020-03-08:
The problem was wire connector , that was thin and can not deliver led power usage