Legion2/CorsairLightingProtocol #115: Lighting Node CORE pin an löten

skadzynski, 2020-03-11:
Hi ich hab ein Lighting Node CORE
[image]

Kann man eine pin leiste an löten 
[image]
Und so verbinden?
[image]

skadzynski, 2020-03-11:
[image]

Legion2, 2020-03-11:
@skadzynski ich verstehe nicht was du da anschließen willst, dem Wiring Diagramm zu folge möchtest du LED Lüfter anschließen. Aber die kann man doch an die anderen 6 Anschlüsse anschließen.

skadzynski, 2020-03-11:
Das Bord hat doch drei unbedegte ausgänge

Ohne pins ich will den rgb hub an einen 
Lighting Node pro anschliesen, wenn es überhaupt geht.

Legion2, 2020-03-11:
> Hi ich hab ein Lighting Node CORE

Auf dem Bild ist ein Fan Hub zu sehen.

Kannst du nochmal genau auflisten was du als Hardware hast, was du davon verbinden willst und was du damit erreichen willst. Bitte unterscheide dabei ob original Corsair Geräte und fake Corsair Geräte.

skadzynski, 2020-03-11:
Ich hab den Lighting Node CORE original im gehäuse verbaut, da sind 3 sp rgb pro dran, hab noch nen 3er pack LL lüfter geholt und hab den rgb fun hub demontiert da ja bei dem 3er pack alles da bei ist, allerdings hab ich jetzt halt 2 verschiedene Lüfter an einem rgb fun hub,
Da dachte ich mir irgend wie muss ich doch den Lighting Node CORE aus dem pc gehäuse noch mit an den Lighting Node pro anschließen können.

skadzynski, 2020-03-11:
Das sind alles original teile

Legion2, 2020-03-11:
Den Lighting Node CORE kann man nicht mit einem Lighting Node pro verbinden. Beide brauchen eine USB connection. Du musst die Verschiedenen Lüfter typen mit verschiedenen Lighting Channels betreiben, der Lighting Node CORE hat nur einen Channel. Da du nur einen Fan Hub hast, kannst du auch nicht beide Lüfter typen über den Lighting Node pro betreiben.

skadzynski, 2020-03-11:
[image]
 Aber kann es sein das die pinout die gleichen sind wie beim pro

Legion2, 2020-03-11:
Ja das kann schon sein, ich habe aber noch nicht in meinen CORE und PRO reingeschaut

skadzynski, 2020-03-11:
Ah ok, hätte ja sein können.
Dank dir trotzdem.

closed 2020-03-11 by Legion2