Legion2/CorsairLightingProtocol #116: CoPro / cLNP Device Error Code 10

AlpineWhite, 2020-03-13:
**Describe the bug** 
When flashing a Leonardo, the device fails to initialize with error code 10, and does not show up in iCue (excepting LS100, which works fine).  Same board works without issue when flashed with LS100 sketch and with device set to LS100.  Issue persists after attempting to manually set deviceID, setting deviceID to 00 00 00 00, and manually setting the serial number to 15 characters (hex range) or 31 characters (same length as genuine LNP).

**To Reproduce**
- Connect arduino
- Select Board CLP Arduino Leonardo
- Select Device Lighting Node PRO or Commander PRO
- Select File > Examples > Corsair Lighting Protocol > LightingNodePRO or CommanderPRO
- Select 'Upload'
- Reset iCue and reboot PC

**Expected behavior**
iCUE sees either a Commander PRO or Lighting Node PRO connected after flash and restart of iCUE services.

    OS: Windows 10
    Board: Clone Pro Micro
    IDE: Arduino IDE
    Version: 1.18.10
    Sketch: LightingNodePRO / CommanderPRO

**Code changes**
None

[image]
[image]

AlpineWhite, 2020-03-13:
Adding SIV link status and USB status
[image]
[image]

Legion2, 2020-03-13:
Sry, but these problems are very difficult to debug, if the LS100 works, why not use it? Do you want to control fans?

The problems looks like an initialization problem caused by some bad memory management.

Legion2, 2020-03-13:
@AlpineWhite Which version of the library you are using? Where did you get the sketches from? (Examples in the Arduino IDE/downloaded from GitHub)

AlpineWhite, 2020-03-14:
Sources for the sketches I used were both the 10.0 release on this project, and the latest release of your CLPboards project,  Your question did give me an idea, and I think I will download v9 (I think I saw some other users with that working in the issues) and give that a try for the LNP.  

I am trying to use 4x LNPs because they 'fit' the LED breakdowns of my components better (8x Phanteks Digital Halos, 3x custom LED strips, 2x reservoirs with 14led each, undecided CPU block [where LED count is a multiple of 4], and 2x GPU waterblocks with an unknown count of LEDs.)  I suppose I could use the LS100s, but not sure if iCUE will try and monitor my display for every connected LS100 (which would be a resource hog due to the resolution).

I do appreciate the work you have put into this project.  It's pretty incredible.

Legion2, 2020-03-14:
Thank you, please keep in mind you should manage the library with the Arduino Library Manager, to make sure it's installed in the correct place.

I found a problem in the CLPboards project and released an update. Maybe it can fix the problem.

AlpineWhite, 2020-03-14:
OK I did a fresh reinstall of Arduino (1.18.12), deleted all Appdata/Documents/Program Files folders prior to reinstall.  

I used library manager to install only the bare necessities and then tried uploading the example sketches using the Boards 1.3 and Protocol .10.  Sadly, same results.

I then tried using library versions back to .8 with the same result.  I went back and removed all versions, and walked through the visual studio method, to the same results.

I then went back to all latest releases, and burned fresh Pro Micro bootloader onto one of the boards. Hit the same roadblock there too.  One constant through all of it was that the USB device shows as type '00' instead of type '03.'  I grabbed iCUE logs but it seems the device is not hitting iCUE at all.

For the time being, I have flashed 3 of the boards with LS100, one set up with 138/30 LEDs and the others as 96/96.  This seems to be working fine at the iCUE level (though I have not tried adding LED strips to the outputs).

[image]

Is the device type (e.g. USB HID device) defined in the code, or is that assigned by the driver?  I do see the same driver being used for all devices (genuine and not).

[image]

Legion2, 2020-03-15:
Corsair devices (LNP, LNC, LS100 and ComPro) are HID devices, so they can use the windows default USB HID device driver. That's the cool thing about USB HID, that's also the cause why you can use USB mouse and keyboard without installing drivers.

What board do you have exactly? In your first post you said it's a Leonardo and Pro Micro clone. In the first post you also said you used the CLP Arduino Leonardo board, but if it's a Pro Micro you should use the CLP SparkFun Pro Micro board. There are many resellers rebranding the SparkFun Pro Micro to something with Pro Micro or Leonardo Pro Micro, so it's important to known if you should use the Arduino Leonardo, SparkFun Pro Micro or some other board definition to correctly upload sketches.

Did you try to upload the LNC sketch, just to see if it works?
From your posts and screenshots I can tell, that the problem is before or in the `setup` method. The USB stack is not correctly initialized. This could be due to duplicate Serial Number, not enough memory, buffer overflow. I don't understand why the LS100 works, because at the USB layer all devices do the same.

AlpineWhite, 2020-03-15:
Board is the [Kookye Pro Micro clone](http://kookye.com/2016/02/22/kookye-pro-micromini-atmega32u4-module-board-for-arduino-leonardo-replace-atmega328/).

The mfg notes to use the Leonardo board definition, but it will upload/accept both.  I tried with the [SparkFun TX/RX LED blink](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/example-1-blinkies), and the SparkFun board definition causes the delay between LEDblinks to be too short.  With Leonardo selected, the delay is an actual 1000ms. That is how I chose the Leonardo as the proper board definition.

Using board definition CLP Leonardo, example LightingNodeCORE sketch, and device "Lighting Node CORE" I also have it show up in iCUE without issue.
[image]

**My Upload Process:**
- Hold reset shorted to ground
- Press Upload in Arduino IDE
- Wait for IDE output to start listing COM ports
- Release reset condition

This made me wonder.  I set board to CLP Leonardo, set device to "Lighting Node PRO" and went through all the sketches.
None of the sketches worked with the device set to Lighting Node PRO.
Keeping board and device the same (CLP Leonardo and "Lighting Node PRO") I uploaded the blink sketch earlier, which worked fine (except serial monitor).

To test your serial theory, I went to a [random hex generator](https://www.browserling.com/tools/random-hex) and had it generate 5 strings of 31 characters, which corresponds with the genuine lighting node pro serial length.

[image]

I uploaded the device ID sketch and verified the device ID is unique (set Board to Arduino Leonardo so I have Serial COM Port):
[image]

I then opened the "Additional Features" sketch and set the serial number to one of the generated hex 'serial' numbers (6A1DF02EBFBAE945A366E082C99024C). Board set back to CLP Leonardo and device set to "Lighting Node PRO."
[image]

After upload, checking usbdeview shows the serial is concatenated to 19 characters, similar to [Cannot add multiple Node PRO #11](https://github.com/Legion2/CorsairLightingProtocol/issues/11)

[image]

Interestingly, the default board serial is the same length as the CommanderPRO (Board > CLP Leonardo, device "Commander PRO", sketch "CommanderPRO" from examples, no edits.) Arduino is highlighted, genuine CoPro is above it:

[image]

ParentID Prefix is not set, which it is for all other genuine Corsair products, no SurpriseRemovalOK flag, and no other observed differences.  Going back and comparing genuine and programmed LNP, same differences are observed:

[image]

Let me know what else I can do to try and help, and I'll give it a shot.

Legion2, 2020-03-15:
The Serial Number length is limited to 20 by Arduino Core. There was a buffer overflow so I added the following code to respect this limit. I also opened an issue at arduino core arduino/ArduinoCore-avr#83.
https://github.com/Legion2/CorsairLightingProtocol/blob/ec2b9ebc5c8a82a70684224bd79e0affd37fb206/src/RawHID.cpp#L168-L172

The short serial number is no problem for iCUE, the serial number is only used by Windows and must be unique.

> This made me wonder. I set board to CLP Leonardo, set device to "Lighting Node PRO" and went through all the sketches.
None of the sketches worked with the device set to Lighting Node PRO.

So you mean it's not related to the sketch but rather a problem with the device definition of "CLP Leonardo" and "Lighting Node PRO" device?

AlpineWhite, 2020-03-15:
> 
> So you mean it's not related to the sketch but rather a problem with the device definition of "CLP Leonardo" and "Lighting Node PRO" device?

Interesting info on serial length for sure.  I do believe it may be a a problem with the definition of the Lighting Node PRO and Commander PRO devices.  
CLP Leonardo + LS100 = Good (Example sketch and blink both run OK)
CLP Leonardo + Lighting Node CORE = Good (Example sketch and blink both run OK)
CLP Leonardo + Lighting Node PRO = Fail (blink runs OK, code execution seems to be unaffected)
CLP Leonardo + Commander PRO = Fail (blink runs OK, code execution seems to be unaffected)

Legion2, 2020-03-15:
The only difference in the definitions are the USB VID and the USB Product Name:
https://github.com/Legion2/CorsairLightingProtocolBoards/blob/6724badcdd39716788c02c93828f052bd7e4fca7/avr/boards.txt#L148-L159

AlpineWhite, 2020-03-15:
I just figured it out.  This has something to do with mixing genuine and non-genuine Corsair components.  I was unable to use CoPro nor LNP because I have genuine versions of both devices.  I do not have any QL fans nor a genuine LS100, so both of those devices (LS100 and LNC) worked.

**Steps to solution:**
-Use SIV to find the address of the USB hub holding the genuine Lighting Node PRO.
-Disable hub. (which is on my Commander PRO)
-Right click on flashed device, select disable, check the box to uninstall device driver software.
-Re-plug flashed device.
-Lighting Node PRO is now ready to use.

I can re-enable the hub, and now see both the genuine and fake Lighting Node PROs. I have not tested a reboot.

When enabled this way, it does not identify itself in USBDeview as a Lighting Node PRO (though it does show in iCUE).
[image]

SIV also shows it quite differently from the genuine LNP:
[image]

I cannot test the same on my CoPro because it runs the pumps and fans in my rig.

In light of this revelation, I now feel comfortable sharing [my quad mount for 4X 'devices'](https://easyeda.com/p.cartwrig/corsairlightingprotocol) on a molex-powered voltage-selectable (per device) bus.  M3 screw holes for mounting on standoffs on the back of a mobo tray or wherever else. Integrated reset jumpers for remote switches for programming/recovery. 470uF cap per LNP(25V), and 100-Ohm resistor per channel for signal stability.  Pinout is [VCC D NC GND] to match mobo header pinout on DRGB/ARGB motherboards.

[image]

I am a bit curious though, do you have any experience mixing unofficial and official controllers?

Legion2, 2020-03-15:
I have a genuine ComPro and genuine LNC and two non-genuine LS100 and they work without any problems. I now also tested a non-genuine LNC and non genuine ComPro and they show up in iCUE but they mess up the ComPro temp sensors and Fan detection of the genuine ComPro.

The non-genuine devices looks different in SIV because, it has the arduino default usb settings, like the serial port, which can't be disabled easily. From my experience I know iCUE uses the HIDSharp library to access the devices and use only use the VID:PID to get all devices. All other properties of the device, like the serial number are ignored. Internally iCUE uses the DeviceID to identify the devices.

I think the "uninstall device driver software" did the job, because maybe there was a special old driver installed for LNP that did not work with the non-genuine LNP.

AlpineWhite, 2020-03-16:
I got a chance to reboot, and can confirm that reloading the drivers has resolved this issue.  Thank you for the incredibly fast support, and insight into the workings.

closed 2020-03-16 by AlpineWhite

Legion2, 2020-03-16:
You are welcome and thank you for sharing your PCB and the LS100 enclosure.