Legion2/CorsairLightingProtocol #125: Question about functionality

julianflapper, 2020-04-23:
Hi,

This project seems really cool and I'm interested in looking at setting this up instead of a Commander Pro.

I'd like to control the speed and lighting of up to 9 PWM / RGB fans. 

If I'd use this project to create a Commander Pro out of an Arduino, would I get the full functionality or is it still recommended to actually buy one and only use this project as an extension?

Legion2, 2020-04-23:
The Commander Pro created with this library is not fully functional see [wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/Commander-PRO).
And like a normal ComPro can only control 6 Fans individually.

julianflapper, 2020-04-23:
Alright, thanks! Awesome project btw :)

closed 2020-04-23 by julianflapper