Legion2/CorsairLightingProtocol #126: Adding stm32f103c8t6 support

NikN52, 2020-04-28:
Dear friend, I am shocked by your project! Could you add support for the stm32f103c8t6 boards or tell me how I can do this? When I try to compile, I get the following message:
```
WARNING: the CorsairLightingProtocol-dev library must run on avr architectures and may not be compatible with your STM32F1 architecture board.
In file included from C: \ Program Files (x86) \ Arduino \ libraries \ CorsairLightingProtocol-dev \ src / CorsairLightingNodePRO.h: 18: 0,

                 from C: \ Program Files (x86) \ Arduino \ libraries \ CorsairLightingProtocol-dev \ examples \ LightingNodePRO \ LightingNodePRO.ino: 16:

C: \ Program Files (x86) \ Arduino \ libraries \ FastLED-3.3.3 / FastLED.h: 14: 21: note: #pragma message: FastLED version 3.003.003

 # pragma message "FastLED version 3.003.003"

                     ^

LightingNodePRO: 22: 1: error: 'CorsairLightingNodePRO' does not name a type

 CorsairLightingNodePRO cLNP;

 ^

C: \ Program Files (x86) \ Arduino \ libraries \ CorsairLightingProtocol-dev \ examples \ LightingNodePRO \ LightingNodePRO.ino: In function 'void setup ()':

LightingNodePRO: 26: 23: error: 'cLNP' was not declared in this scope

  auto ledController = cLNP.getFastLEDController ();

                       ^

C: \ Program Files (x86) \ Arduino \ libraries \ CorsairLightingProtocol-dev \ examples \ LightingNodePRO \ LightingNodePRO.ino: In function 'void loop ()':

LightingNodePRO: 32: 2: error: 'cLNP' was not declared in this scope

  cLNP.update ();

  ^

exit status 1
'CorsairLightingNodePRO' does not name a type
```

Thank you!

Legion2, 2020-04-28:
@NikN52 Can you provide documentation for the USB HID API of the stm32f103c8t6. See also #93

carlosvcsd, 2020-06-21:
Please, can you suport boards like Esp8266 ? I have a Wemos D1 mini and i would like to test this in my board.

Legion2, 2020-06-21:
@carlosvcsd Have a look at #109 for the reason why ESP8266 is not supported.

closed 2020-08-07 by stale[bot]