Legion2/CorsairLightingProtocol #127: Light Glitch

MajorLOLGaming, 2020-05-02:
**Describe the problem**
Light effects has some stutters and random colored flashes. (Only happens on the board using CLP)
I suspect this may be due to cable length or interference from the 6ish cables with 240v running in them.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Leonardo Pro Micro
 - IDE: Arduino IDE
 - Version: 0.10.0
 - Sketch: LightingNodePRO

**Code changes**
Did you change the code of the library or the sketch?
No.

MajorLOLGaming, 2020-05-02:
Maybe adding the string:
`#define FASTLED_ALLOW_INTERRUPTS 0`
would do some changes? (Source(s): [1](https://github.com/FastLED/FastLED/issues/306#issuecomment-219868579) [2](https://github.com/FastLED/FastLED/wiki/Interrupt-problems))
I will come back with an update as to if it worked.

Legion2, 2020-05-02:
Which LED chipset are you using? I'm using WS2812B for one year now and have no problems.
When I had random light flashes, the LEDs were broken.

MajorLOLGaming, 2020-05-02:
I am also using WS2812B, but when i use some other driver, it has no flashes.

MajorLOLGaming, 2020-05-02:
im guessing a possible reason could also be poor connection between the board and leds? Or from power supply to leds?

Legion2, 2020-05-02:
How do you power the LEDs?

Legion2, 2020-05-02:
Did you connect GND of the Arduino with the GND on the power supply?

MajorLOLGaming, 2020-05-02:
I power the leds from an 8 amp 5v psu, and i have a common ground between the leds and the board.

MajorLOLGaming, 2020-05-02:
also, my leds are running off the psu, while the board is running on usb power

MajorLOLGaming, 2020-05-02:
i will try and solder everything directly on it.

MajorLOLGaming, 2020-05-02:
Hmm... I think the led data cables were just too long. It seems to have either been fixed, or stopped being so extreme.

MajorLOLGaming, 2020-05-02:
Hmm... It's weird. Every time i try and mount it near my other cables, it start to flicker again...

Legion2, 2020-05-03:
This doesn't sound like a problem with the implementation of this library.

closed 2020-05-05 by MajorLOLGaming