Legion2/CorsairLightingProtocol #128: CommanderPro FAN NOISE

CorsairGuy, 2020-05-07:
**Describe the problem**
Hello everyone. I am designing a circuit board for CommanderPro and have encountered several problems:
1) Power supply for Corsair fans - 12V. Is the PWM signal also 12V or 5V? I know that there are 12V power / 5V pwm fans.
2) I tried connecting my Arduino Leonardo to pin 10 (16bit PWM) to the fan and heard a loud squeak. I tried to find on the Internet what was the reason - as it says, for PWM FANS you need a 25khz PWM signal. so that he is inaudible to man. Also on the Internet I found an interesting code on how to do this, but you need a 16bit PWM output. Atmega 32u4 there are 2 of them - 9 and 10 pins. Below I have provided code that does this.

`// PWM output @ 25 kHz, only on pins 9 and 10.
// Output value should be between 0 and 320, inclusive.
void analogWrite25k(int pin, int value)
{
    switch (pin) {
        case 9:
            OCR1A = value;
            break;
        case 10:
            OCR1B = value;
            break;
        default:
            // no other pin will work
            break;
    }
}

void setup()
{
    // Configure Timer 1 for PWM @ 25 kHz.
    TCCR1A = 0;           // undo the configuration done by...
    TCCR1B = 0;           // ...the Arduino core library
    TCNT1  = 0;           // reset timer
    TCCR1A = _BV(COM1A1)  // non-inverted PWM on ch. A
           | _BV(COM1B1)  // same on ch; B
           | _BV(WGM11);  // mode 10: ph. correct PWM, TOP = ICR1
    TCCR1B = _BV(WGM13)   // ditto
           | _BV(CS10);   // prescaler = 1
    ICR1   = 320;         // TOP = 320

    // Set the PWM pins as output.
    pinMode( 9, OUTPUT);
    pinMode(10, OUTPUT);
}

void loop()
{
    // Just an example:
    analogWrite25k( 9, 110);
    analogWrite25k(10, 210);
    for (;;) ;  // infinite loop
}`

Then, I started looking for something similar in your library. In the file CorsairLightingProtocol / examples / CommanderPRO / PWMFan.cpp I found the lines that I cited below. ** Please tell me what should I do ** to combine the code above and the code from your library? I really need your help! (sorry for the bold, I wanted you to see this among this pile of text and sorry for my English, it's terrible :( )

I can also say that I have already designed a circuit board for the LS100. If necessary, I can upload it to the Internet so that someone can create a “miracle”. Now I'm designing a body for her.

`PWMFan::PWMFan(uint8_t pwmPin, uint16_t minRPM, uint16_t maxRPM) : pwmPin(pwmPin), minRPM(minRPM), maxRPM(maxRPM) {
	pinMode(pwmPin, OUTPUT);
	analogWrite(pwmPin, 0);
	switch (digitalPinToTimer(pwmPin)) {
		case TIMER0B: /* 3 */
#ifdef DEBUG
			Serial.println(F("Pin not supported as PWM fan pin"));
			Serial.println(F("We don't want to mess up Arduino time functions"));
#endif  // DEBUG
			break;
		case TIMER3A: /* 5 */
			TCCR3B = (TCCR3B & B11111000) | 0x01;
			break;
		case TIMER4D: /* 6 */
			// PLLFRQ = (PLLFRQ & B11001111) | (0x03 << PLLTM0);
			TCCR4B = (TCCR4B & B11110000) | 0x01;
			break;
		case TIMER1A: /* 9 */
			TCCR1B = (TCCR1B & B11111000) | 0x01;
			break;
		case TIMER1B: /* 10 */
			TCCR1B = (TCCR1B & B11111000) | 0x01;
			break;
		default:
#ifdef DEBUG
			Serial.println(F("Pin not supported as PWM fan pin"));
#endif  // DEBUG
			break;
	}
}`

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo
 - IDE: Arduino IDE
 - Version: 1.8.12
 - Sketch: CommanderPRO/PWMFan.cpp

Legion2, 2020-05-07:
The code you found in `PWMFan.cpp` should already set the timer to maximum speed `0x01` (no prescaling)
```c++
case TIMER1A: /* 9 */
	TCCR1B = (TCCR1B & B11111000) | 0x01;
	break;
case TIMER1B: /* 10 */
	TCCR1B = (TCCR1B & B11111000) | 0x01;
	break;
```
But I don't understand all the other options:
```c++
TCNT1  = 0;           // reset timer
TCCR1A = _BV(COM1A1)  // non-inverted PWM on ch. A
       | _BV(COM1B1)  // same on ch; B
       | _BV(WGM11);  // mode 10: ph. correct PWM, TOP = ICR1

ICR1   = 320;         // TOP = 320
```

CorsairGuy, 2020-05-07:
> The code you found in `PWMFan.cpp` should already set the timer to maximum speed `0x01` (no prescaling)
>
> ```c++
> case TIMER1A: /* 9 */
> 	TCCR1B = (TCCR1B & B11111000) | 0x01;
> 	break;
> case TIMER1B: /* [...]

I think this is the problem. Please, could you just insert this code instead of yours so that it would be correct. Thx

Legion2, 2020-05-07:
I don't insert code whichI don't understand, if you can explain what it does, I will add it. You can test the new code simply by calling it after you created the `PWMFan` objects. This will effectively overwrite my the PWM settings.

CorsairGuy, 2020-05-08:
Хорошо, я попытаюсь экспериментировать сегодня вечером. У меня есть еще один вопрос связанный с CommanderPro. http://www.skillbank.co.uk/arduino/measure.htm согласно этой статье, Atmega 32u4 может измерять аналоговыми входами напряжение от 0 до 5V. В файле **CorsairLightingProtocol/src/CorsairLightingProtocolConstants.h** я нашел следующие строки `#define READ_VOLTAGE_VALUE 0x12` и хотел бы узнать, как я могу отправить полученное напряжение по USB в iCUE чтобы оно корректно отображалось?

CorsairGuy, 2020-05-08:
By default the analog inputs measure from ground to 5 volts, though is it possible to change the upper end of their range using the AREF pin and the analogReference() function.

CorsairGuy, 2020-05-08:
As I understand it, I need to find out the variable and some bits, as well as append a call to this function in some file and send it to USB, please tell me how can I do this?

Legion2, 2020-05-08:
you must return the voltage values here then they will be shown in iCUE https://github.com/Legion2/CorsairLightingProtocol/blob/ae284f25bf4daf13fd29f9f97f40005e9cb672fe/examples/CommanderPRO/ThermistorTemperatureController.cpp#L53-L63

CorsairGuy, 2020-05-08:
This is amazing! Thank you, today I will try to do it!

Legion2, 2020-05-08:
The problem that I see and why I haven't implemented jet, is that the arduino always measure Voltage by comparing to a reference voltage (AREF pin). So you have to make sure that this reference voltage is a fixed value and the only such constant source I can think of in this context is the 5V of the PSU itself. But this will make no sense, because we want to measure the 5V of the PSU, so the Arduino compares the 5V from PSU at AREF pin with the 5V from the PSU at the analog pin and will always return 100%. So this measurement is useless. Even if the voltage drop to 4.8V the reference also drop to 4.8V.

Legion2, 2020-05-08:
I should add this to the documentation

CorsairGuy, 2020-05-08:
But what if we consider the standard 12V and consider 5V and 3V3 relative to it?

CorsairGuy, 2020-05-08:
How do everyone else use CommanderPro? Doesn't this squeak bother anyone? Or is a ready-made solution already invented?

Legion2, 2020-05-08:
The main focus of this library is lighting which is simple with FastLED, controlling fans is complicated because of the many pins and no library that manage different fan types (#27). Also for some fan extra hardware like mosfet is required. https://github.com/TylerSeiford/CorsairController is a project which build a hardware controller. The features of the Commander PRO are not complete https://github.com/Legion2/CorsairLightingProtocol/wiki/Commander-PRO.

The issue with the noise was reported before #75.

CorsairGuy, 2020-05-08:
Yes, I read this report, and therefore immediately tried to connect to Leonardo. I will try to study the ComPro revision you pointed out. If I can get rid of the noise, I will have to modify your library because I want to get all the functions of the original controller.
As I understand it, 32u4 cannot measure 12V and will break if I try to do this. I guess I should use something else to measure. The USB hub can be implemented on another processor.

CorsairGuy, 2020-05-08:
If it’s easy for you, could you take a picture of the ComPro PCB from both sides, please.
Also photos of the LS100 printed circuit board would be useful to me, if you can - please do it for me.

CorsairGuy, 2020-05-08:
If anyone is interested in my project, this is what the PCB LS100 looks like
[image]

Legion2, 2020-05-08:
I created the library without owning any of corsair devices (only a keyboard), I reverse engineered it only using [this post](https://github.com/audiohacked/OpenCorsairLink/issues/70) and trial and error. So I can't provide you with the PCB information.

CorsairGuy, 2020-05-08:
Do you want me to go your way? You mentioned that you have the original 2 pieces LS100 and CommanderPro

Legion2, 2020-05-08:
I currently don't have a real LS100 only a ComPro. I have multiple Arduino LS100.

CorsairGuy, 2020-05-08:
Do you want me to go your way?

CorsairGuy, 2020-05-08:
I did it! I added the lines of code that I indicated above and everything works super! You also need a mosfet and 2 resistors. Now there is no noise coming out, and the PWM signal from the arduino from 5V turns into 12V. Why improve your library, don’t add this, but if someone wants to do the same as me, I can tell you how to do it best.

Sorry, but the creator of this library does not want any outside interference.

CorsairGuy, 2020-05-08:
Also, my engineer friend told me how to correctly and easily measure voltage, including 5V and 12V. Later I will tell you how to do it ..

Legion2, 2020-05-08:
> I did it! I added the lines of code that I indicated above and everything works super! You also need a mosfet and 2 resistors. Now there is no noise coming out, and the PWM signal from the arduino [...]

Can you please open a Pull Request with your changes and then I will review them. The Problem with your initial suggestion was, that you wanted me to just "insert" the "correct" code without testing it.

CorsairGuy, 2020-05-08:
Give me 10 minutes please

CorsairGuy, 2020-05-08:
How i can open a Pull Request?

CorsairGuy, 2020-05-08:
I unfortunately do not have time to study Github, but I can say what I did or send you a file.

CorsairGuy, 2020-05-08:
I seem to have done what you need.

CorsairGuy, 2020-05-08:
For correct measurements, we need a reference voltage source. More - I can tell you in a couple of days when I do it in my project.

CorsairGuy, 2020-05-08:
> Also for some fan extra hardware like mosfet is required. https://github.com/TylerSeiford/CorsairController is a project which build a hardware controller.

Я просмотрел весь код, и не нашел никаких отличий от вашего кода, видимо он думал что осилит это, но когда осознал что очень сложно, забил на это дело.
Можно сказать что я уже сделал для вашего проекта больше, чем он.

TylerSeiford, 2020-05-08:
My hardware project was a custom circuit board with an ATmega32U4, MosFETs, and accompanying support hardware to replicate and extend the functionality of the CommanderPRO.  My design has a single 12V non-addressable LED strip connector, and two 12V fan connectors that can be switched between 3 and 4 pin modes using a jumper.  In other versions I have support for more or less fans, addressable LED headers (both of the corsair 12V variety and standardized 5V variety), temperature inputs, and more.  The limiting factors are the number of pins available on the ATmega32U4 (specifically interrupt and PWM capable).

Depending on the type of fan you use, the hardware requirements vary.  To control a non-PWM 12V fan (3-pin) you need to have a mosFET to PWM the power to the fan, and for a PWM 12V fan you only need to wire power and PWM to the fan.

CorsairGuy, 2020-05-09:
And yet you didn’t realize anything conceived, except for your useless printed circuit board ..

CorsairGuy, 2020-05-09:
hey friend, you can't just take and process the tach signal. I'm not sure that this is possible, but a huge harness is needed. it's not worth it. there is no need to use the tachometer.

if you really want to measure this, you can use the output on the motherboard and design your PCB as a repeater of this signal for each cooler. Cosair did this in his repeater. just find it.
this is a good decision. but I don’t want to control my fans through third-party software. I want to use only iCUE.

CorsairGuy, 2020-05-09:
**Disclaimer**. I'm sorry that I write so much, I just want someone to do the same and not come across these problems. therefore, I try to paint each step in great detail. I hope you appreciate it.

CorsairGuy, 2020-05-09:
@Legion2 http://r6500.blogspot.com/2014/12/fast-pwm-on-arduino-leonardo.html please read this article. Perhaps you can get from the fourth timer 48kHz or 96kHz. And we will have 4 individually controlled fans.

CorsairGuy, 2020-05-09:
On the printed circuit board, I plan to use the MAX31790 to control the fans, it is able to control 6 fans and receive a signal from their tachograph. How can I add your library to add a TACH signal and send it to iCUE?

Legion2, 2020-05-09:
@CorsairGuy Please keep this discussion friendly and respectful. @TylerSeiford was the first who contributed a PR to my project and help me to write and test the Commander PRO code.

> UPD: tl431 - most common voltage reference.You can find the wiring diagram and solder it to your arduino. I’m developing a printed circuit board, which you can later order from any manufacturer of 5 pieces for $ 5 (excluding delivery, with delivery to Russia $ 10)

> UPD: since it works through the I2C protocol, I will need your help in changing your library, can I get it?

@CorsairGuy you use many different hardware components in your project that requires very specific software. You want this software to be included in this library, but it's out of scope of this library to support your very specific software/hardware needs. This library is un-opinionated about which hardware you should use. Everything is open source, feel free to use and customize the library for your project. But I have no time to write specific code for your project.

CorsairGuy, 2020-05-10:
You can set another timer and get 4 controlled fans. I indicated the link above. You will not do it?

TylerSeiford, 2020-05-12:
> And yet you didn’t realize anything conceived, except for your useless printed circuit board ..

My circuit board and accompanying hardware have given me the closest thing to a custom CommanderPro as can be supported by the ATmega32U4 without either fancy timer tricks or external chips (both of which I considered but decided against using).

> you can't just take and process the tach signal

Using the interrupt pins of the arduino you can actually measure the tachometer input from fans pretty accurately but I decided against implementing it into my project merely on the basis of the amount of time required.

CorsairGuy, 2020-05-12:
yes not a question brother. however, I will do more functionality than the original ComPro

CorsairGuy, 2020-05-12:
I can show you my sketch, which perhaps has nothing to do with the final version ..
[image]

CorsairGuy, 2020-05-12:
Unlike the original, it has 6 J_USB and 6 J_FAN_LED. Plus, one port is occupied by the ATmega 32U4, which is why the MicroUSB wire is used, and not as in the original DUPONT with 2 USB ports.

closed 2020-06-28 by stale[bot]