Legion2/CorsairLightingProtocol #142: Increase brightness when using ls100 video lighting?

tchesket, 2020-06-04:
**Describe the problem**
Hi, I was wondering if it is possible to increase the overall brightness of the LEDs when using the ambient backlight effect. The leds seem really dim compared to when directly controlled. Is there a way to apply some kind of brightness scaling to the leds before each update, or to set a minimum brightness perhaps? Thank you.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo
 - IDE: Arduino IDE
 - Version: 1.8.33
 - Sketch: AmbientBacklight

**Code changes**
Did you changed the code of the library or the sketch?
Added CLP::scale function to onUpdateHook

Legion2, 2020-06-04:
By default all iCUE software playback effects have a maximum brightness of 50% only, which can't be changed in iCUE. You can change this in the `onUpdateHook` by multiplying all RGB values by 2 before applying `gammaCorrection`.

This workaround only works in Software Playback mode, in HW the default brightness is 100% and then all colors are broken because of the scaling.

tchesket, 2020-06-04:
Could you give me an example for the code I would use to multiply the rgb values? Sorry I'm really new to all this

Legion2, 2020-06-04:
I can do this later today

tchesket, 2020-06-04:
Okay awesome, thank you

Legion2, 2020-06-04:
@tchesket Have a look at the new `fixIcueBrightness` function https://github.com/Legion2/CorsairLightingProtocol/pull/143/files#diff-df06cb0ecab891d52d16b3deb8423144R100

tchesket, 2020-06-04:
This seems to have worked, thanks for doing that :)

Sanbay, 2020-06-07:
i just changed FastLEDController.cpp(v 0.12.0)  line 440 (`case ChannelMode::SoftwarePlayback:`)

`data.leds[i] = CRGB(data.valuesBuffer[0][i], data.valuesBuffer[1][i], data.valuesBuffer[2][i]);`

to

`data.leds[i] = CRGB(data.valuesBuffer[0][i]*2, data.valuesBuffer[1][i]*2, data.valuesBuffer[2][i]*2);`

thats fixes LS100 software brightness issue

tchesket, 2020-06-07:
Ah okay. So if I download everything again with the changes you made, I can
remove the fixIcueBrightness line and it should work?

Sanbay, 2020-06-08:
Discard that, download latest version and use `fixIcueBrightness` function

```
ledController.onUpdateHook(0, [](){
CLP::fixIcueBrightness(&ledController, 0);
});
```

tchesket, 2020-06-08:
Oh okay, that's what I'm already using

PR #143 (2020-06-08): added fixIcueBrightness util function
The function `fixIcueBrightness` doubles the received RGB values so they are in the range (0 - 254) close #142

TODO:
- [x] add this to readme
commit 445d5bc5 (2020-06-08): added fixIcueBrightness util function close #142
closed 2020-06-08 by Legion2