Legion2/CorsairLightingProtocol #145: Downsizing with scale function

Sanbay, 2020-06-08:
Hi
Im not sure its a bug or i make mistake but scale function looks doesn't works well on downsizing.

I have 58 led total ambient lighting setup (9 sides, 20 leds up an down) when im using scale function getting very weird result.

`CLP::scale(&ledController, 1, 58);`

``void CLP::scale(FastLEDController* controller, uint8_t channelIndex, int scaleToSize) {
	auto leds = controller->getLEDs(channelIndex);
	const float scaleFactor = (float)controller->getLEDCount(channelIndex) / scaleToSize;
	for (int ledIndex = scaleToSize - 1; ledIndex >= 0; ledIndex--) {
		leds[ledIndex] = leds[lround(ledIndex * scaleFactor)];
	}
}``

i checked the scale function for understands how its works. Looks like function starts latest leds on actual setup and copies values corsair default setup by multiplier.

Lets say i have 48 led setup and default ambient led count is 96, if i use scale function scalefactor gonna be 2.00 and this function gonna start marge values like this

leds[47] = leds[94];
leds[46] = leds[92];
leds[45] = leds[90];
leds[44] = leds[88];

everythings ok until when multiplied number equal or less then actual led length.

leds[23] = leds[46]; (this value already copied from leds[92])
leds[22] = leds[44]; (this value already copied from leds[90])

leds[11] = leds[22]; (this value already copied from leds[44])

so, when ,cue changed colors of 88th led   led[44] ,led[22] and led[11] gets same colors vs vs.

Legion2, 2020-06-08:
@Sanbay can you test #147

Sanbay, 2020-06-09:
Awesome!! Thanks  :tada:

PR #147 (2020-06-09): fixed scale function when scaling down
 fix #145
commit 3d122e50 (2020-06-09): fixed scale function when scaling down fix #145
closed 2020-06-09 by Legion2