Legion2/CorsairLightingProtocol #149: Wireless RGB

Legion2, 2020-06-18:
There were multiple requests on wireless RGB with CLP, if this is even possible and how to do it. Because CLP is modular it's simple to use it with existing wireless technologies. It is only required that the wireless technologies can reliably transmit (bidirectional) data with an adequate bandwidth. The wireless connection is only used as a bridge between a USB-HID compatible Arduino and remote device.

Two approaches are possible to implement wireless RGB with CLP:
1. CLP runs on the USB-HID compatible Arduino and sends the already processed RGB data via wireless connection to remote device which only have to use for instance FastLED to display them on LED Strips.
2. On the USB-HID compatible Arduino only a small firmware bridge is running which sends the raw CLP Commands via wireless connection to a remote device.
On the remote device CLP is running with an adapter for the wireless connection.

A more detailed description should be added to the Wiki.

TylerSeiford, 2020-06-24:
It would also be nice to be able to use I2C or SPI for this, as you could then use something like the Mega connected over I2C for the I/O.

Legion2, 2020-08-06:
@TylerSeiford you mean you have a 32u4 board connected via SPI or I2C to a mega?

TylerSeiford, 2020-08-06:
Yes, such that you could use the mega (or any similar arduino) to expand the IO capabilities of the 32u4 (for example, to enable more interrupt capable pins for fan tachometer measurements).

Legion2, 2020-08-07:
On the mega you can use the HoodLoader 2 bootloader, and the internal hardware Serial connection. I don't really see an advantage for using SPI over Serial. SPI only send bytes and the synchronization for sending and receiving must be handled on top. I2C have a acknowledge mechanism but by default only a 32 byte buffer. I also think both need interrupts, so may not work correctly with FastLED.