Legion2/CorsairLightingProtocol #150: LS100

skadzynski, 2020-06-21:
Hi
Ich Habe das problem das ich in icue bei dem ls100 wenn ich auf einem kanal nur ws2812b dran hab und auf RGBstreifen stelle das die sich nicht einzelnt steuern lassen, ist das normal?

Legion2, 2020-06-21:
Wenn du in iCUE einen Effekt hinzufügst kannst du die LEDs einzeln anklicken und dadurch den effekt nur auf diese anwenden. Man kann auch mehrere LEDs für einen Effekt auswählen wenn man die Maus gedrückt hält während man einen größeren bereich auswählt. Falls das nicht verständlich ist, schau dir mal tutorials zu iCUE an, dort wird das im Video gezeigt.

skadzynski, 2020-06-22:
Ja ich weiß  wie man makiert😉
Nur funktioniert es nicht wenn ich das mache, die leds gehen dann aus und ich kann dann nix mehr einstellen, erst wenn ich den Effekt wieder lösche und dann leuten beide langen streifen (27 ws2812b) gleich, sobald ich was markiere gehen sie aus.

Legion2, 2020-06-22:
Welcher Version von CLP nutzt du und hast du was an dem Arduino sketch geändert?

skadzynski, 2020-06-22:
Die aktuelle eigentlich, an der ino hab ich nur die RGB Reihenfolge geändert, wenn ich zu hause bin guck ich nach der genauen version.

Legion2, 2020-06-22:
Welche iCUE version hast du? Ich verstehe das problem aktuell noch nicht, kannst du Screenshots machen damit ich das besser verstehe.

skadzynski, 2020-06-22:
Ich weiß nicht was es war, hab es grad fotografieren wollen, und plötzlich funktioniert es.
Dann hat es sich ja geklärt, trotzdem danke.👍

closed 2020-06-22 by Legion2