Legion2/CorsairLightingProtocol #151: OpenRGB Support

CalcProgrammer1, 2020-06-25:
**Describe the bug**
I'm the developer of [OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB) and my project supports the real Lighting Node devices.  I'd like to make sure it works with CorsairLightingProtocol as I would like to be able to recommend an Arduino running this project as a recommended controller for OpenRGB users.

Unfortunately, OpenRGB has trouble controlling an Arduino Pro Micro running CorsairLightingProtocol as a Lighting Node Pro.  A real Lighting Node Pro works correctly.  It may be that my implementation of the Corsair protocol doesn't exactly match iCue, but my code does work on a real Lighting Node Pro.

OpenRGB sends a keepalive (apply) message every 5 seconds as a real Lighting Node device will revert to the saved hardware effect mode after 20 seconds of bus inactivity.  It seems like CorsairLightingProtocol is applying changes on this keepalive rather than on the actual color update message.  After a few tries, it becomes unresponsive.

**What works?**
OpenRGB successfully detects my Arduino as a Lighting Node Pro.  I flashed the Lighting Node CORE sketch so I'm not sure why.  It shows up in Zadig/Device Manager/lsusb as a Pro.  Firmware version is correctly reported.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Arduino as Lighting Node Pro
2. Start OpenRGB
3. Resize channel 1 to number of LEDs (15 tested)
4. Click a color and Set Device

**Expected behavior**
Lights immediately change to selected color

Effect modes work

**Actual behavior**
Lights don't immediately change, sometimes apply set color on the 5s keepalive command.

Effect modes turn off the LEDs

**System:**
 - OS: Windows 10 and Debian Linux
 - Board: Arduino Pro Micro
 - IDE: Arduino IDE
 - Version: Latest
 - Sketch: LightingNodeCORE

**Code changes**
None

Legion2, 2020-06-25:
I update the leds on the command `0x33` (`WRITE_LED_TRIGGER`) which is send by iCUE after a completed update to the configuration, so it guarantees a consistent state (some state is persisted on the Arduino). This is especially important with software playback, because the different colors (red, blue, green) are updated in a sequence of commands, so I can only update the leds after the sequence is complete. And currently this is indicated by the command 0x33. https://github.com/Legion2/CorsairLightingProtocol/blob/5d9594cdbc049c77699276960848117e601ea20b/src/LEDController.cpp#L23-L26

> After a few tries, it becomes unresponsive.

What do you send as keepalive (apply) messages? You mean after some the keepalive messages the arduino becomes unresponsive?

> Here is my protocol documentation.

The "Device Types" are actually the number of LEDs of that device, at least I have implemented it that way and it works.
I called the "Effect Configuration" `LEDGroup`:
https://github.com/Legion2/CorsairLightingProtocol/blob/5d9594cdbc049c77699276960848117e601ea20b/src/LEDController.cpp#L76-L95

CalcProgrammer1, 2020-06-25:
My keepalive loop is a trigger (0x33, I called this CORSAIR_LIGHTING_NODE_PACKET_ID_COMMIT) message sent every 5 seconds.  I also send a commit after every command.

SendCommit: https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/CorsairLightingNodeController/CorsairLightingNodeController.cpp#L258

My Effect Configuration function, it looks like my structure lines up with yours:

CORSAIR_LIGHTING_NODE_PACKET_ID_EFFECT_CONFIG    = 0x35,     /* Effect mode configuration packet     */

https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/CorsairLightingNodeController/CorsairLightingNodeController.cpp#L303

The *10 on the count is probably an error.  I'm going to remove it.

Legion2, 2020-06-25:
> The *10 on the count is probably an error. I'm going to remove it.

Don't remove it, it's the offset on the led strip for the effect. 10 is correct in case of corsair 10 LEDs strip, it should be the sum of all previous `led_type`. I use `usb_buf[0x02]` as offset and `usb_buf[0x03]` as length on the led strip for the effect and this works with iCUE(only uses offset 0) and SIV which can set multiple effects on a channel.

Legion2, 2020-06-25:
I can think of two possible problems (beside the `*10`) on my side.
1. There is a problem when 0x33 is received but there is nothing to commit
2. The keepalive packets are mixed with a sequence of other critical commands and cause some problem when committed in a dirty state.

I will look into these over the weekend.

CalcProgrammer1, 2020-06-26:
I'm going to continue experimenting this weekend as well.  I added some sleeps between packet sends and that seems to have made it a bit more reliable, but it's only controlling the red channel in direct mode.  I haven't messed with the effect modes (built-in effects) yet.

CalcProgrammer1, 2020-06-26:
I uncommented the save and it is still working fine.  The effect I set with OpenRGB is saved and starts back up if I unplug and reconnect it.

My guess is that there's something going on with the writes.  On the real LNP I don't need to do HID reads after each write, and in fact doing so seems to cause slowdowns and stalls.

CalcProgrammer1, 2020-06-26:
Pushed a change to OpenRGB that gets it working with CLP as long as the 64->16 reply size change is implemented.

closed 2020-06-29 by Legion2