Legion2/CorsairLightingProtocol #161: Nexus bar

skadzynski, 2020-07-25:
Hi
Ob es möglich ist das corsair nexus Display nach zu bauen und über ein 3 5zoll Touch Display laufen zulassen?

Legion2, 2020-07-25:
Ich hab da auch schon drüber nachgedacht, aber ich glaube dass die Nexus bar nicht das selbe USB Protokoll verwendet wie die LED Controller von Corsair. Denn die Nexus bar hat eigentlich nichts gemeinsam mit den LED Controllern und daher müsste man beim Reverse engineering von vorne beginnen und ohne selbst eine Nexus bar zu besitzen oder informationen über das USB Protokoll wird es schwierig diese nachzubauen.

skadzynski, 2020-07-25:
Ok. Ich guck mal ob ich eine organisieren kann.

theorgBigBoss, 2020-07-28:
Hi
i have this question too
but i dont undrestand your language

Legion2, 2020-07-28:
@theorgBigBoss here a short translation:

@skadzynski ask if CLP can be used to build a NEXUS by yourself with a touch display.

I then responded that NEXUS probably use another USB protocol as it has nothing in common with the LED controllers. So to reverse engineer the NEXUS protocol I would need access to a NEXUS or at least have some basic information like USB PID.

@skadzynski tries to get one

Legion2, 2020-07-29:
USB ids of iCUE NEXUS:
VID: `1b1c` 
PID: `1b8e`

closed 2020-09-24 by stale[bot]