Legion2/CorsairLightingProtocol #163: Integrate several leds in the base

benoit0727, 2020-08-01:
**Describe the problem**
Hello, sorry for my translated English, I am French, my question is simple, is it possible to design several LEDs for the base, because I made the model of the towers available on thingiverse, unfortunately the led in the base does not light up enough, if you have any idea how to fix this eg have the first four leds work at the same time, otherwise everything works great! Thank you

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Micro Pro
 - IDE: Arduino IDO
 - Version: 1.8.13
 - Sketch: LT100

Legion2, 2020-08-01:
How many leds are in the base? In iCUE they are represented as a single LED. So you would have to use the first LED value from iCUE to control all the physical LEDs of the base. The [Scale separate segments of a channel](https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel) wiki explains how to scale segments. You would use it to define two segments for each tower (base and strip). The base is only one led in iCUE and is scaled to the number of leds in the physical base. The strip of the LT100 is not scaled.

benoit0727, 2020-08-05:
Hi, i tested with the piece of code you sent me, but i don't get a good result, the first led on icue turns me on two physical leds, the second led icue turns me on three physical leds, and after it turns on one by one correctly, I may have made a mistake in the code. Here is the piece of code :

void setup() {
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 108);
	ledController.addLEDs(0, ledsChannel1, 108);
  ledController.onUpdateHook(0, []() {
  CLP::SegmentScaling segments[8] = {{1, 4}, {26, 26}, {1, 4}, {26, 26},{1, 4}, {26, 26}, {1, 4}, {26, 26}};
  CLP::scaleSegments(&ledController, 0, segments, 4);
});
}

Legion2, 2020-08-05:
Which version of this library (CorsairLightingProtocol) are you using?

benoit0727, 2020-08-05:
0.14.0

Legion2, 2020-08-05:
@benoit0727 The code looks good, only in the last line there it should be 8 instead of 4  `CLP::scaleSegments(&ledController, 0, segments, 8);`. I can reproduce your results and need to have a look at the code. I think there is a bug. I'll write you here when I know more.

benoit0727, 2020-08-05:
ok no problem, thank you very much, i will give you the modified code

`#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2

CRGB ledsChannel1[120];

CorsairLightingFirmware firmware = corsairLT100Firmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 120);
	ledController.addLEDs(0, ledsChannel1, 120);
  ledController.onUpdateHook(0, []() {
  CLP::SegmentScaling segments[8] = {{1, 4}, {26, 26}, {1, 4}, {26, 26},{1, 4}, {26, 26}, {1, 4}, {26, 26}};
  CLP::scaleSegments(&ledController, 0, segments, 8);
});
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}`

Legion2, 2020-08-05:
@benoit0727 for multiline code use three ``` and a line break at the beginning and the end:
https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks

closed 2020-08-05 by Legion2

Legion2, 2020-08-05:
@benoit0727 The problem should now be fixed, you just need to update the library to the latest version.

benoit0727, 2020-08-05:
Yeah it works ! Thank you very much!

commit 8fa7aab5 (2020-08-05): corrected incorrect rounding when scaling fix #163
PR #165 (2020-08-05): Added Unit tests
Found a rounding bug in the scaling functions which was resolved with the help of the tests. fixed #163

All the unit tests are in one Sketch and can be uploaded to an Arduino and the results are displayed via the Serial Monitor.
close #144