Legion2/CorsairLightingProtocol #164: Scaling and physical leds.

Justdigit, 2020-08-04:
I need some help with scaling please.

I have 6 fans with 15 controled leds actually there are more leds but there are some parallel soldered.

I have tried to modify the code with scaling. But each time there is one led "missing" in the fan. So animations looks odd. 

Can you explain bit more how to do.?

Legion2, 2020-08-05:
Could you send me the code you used, then I can better explain it to you.

Justdigit, 2020-08-06:
I have actually more fans but this is just for testing three fans.

45 are the actual led segments. 15 per fan....they are hooked up with digital in/out.

and 36 is the Corsair HD fan simulation, 12 per fan.

`/*
   Copyright 2019 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[45];
CRGB ledsChannel2[96];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 45);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 96);
	ledController.addLEDs(0, ledsChannel1, 36);
	ledController.addLEDs(1, ledsChannel2, 96);
  ledController.onUpdateHook(0, []() {
    CLP::repeat(&ledController, 0, 2);
  });
  ledController.onUpdateHook(1, []() {
    CLP::scale(&ledController, 1, 45);
  });
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}`

Legion2, 2020-08-06:
When there is one led "missing" in each Fan and the animations are shifted because of the missing led, try to scale to 16 LEDs for each fan. This should add the "missing" LEDs. Please also update this library to version 0.14.1, I fixed bug in the scaling.

When scaling up you must keep in mind some of the iCUE leds are duplicated to fill the missing information. You can use the LL Fan type in iCUE for 16 LEDs per fan.

Justdigit, 2020-08-06:
> When there is one led "missing" in each Fan and the animations are shifted because of the missing led, try to scale to 16 LEDs for each fan. This should add the "missing" LEDs. Please also update this library to version 0.14.1, I fixed a bug in the scaling.
> 
> When scaling up you must keep in mind some of the iCUE leds are duplicated to fill the missing information. You can use the LL Fan type in iCUE for 16 LEDs per fan.

It's working when I set it with one fan. 15 leds... But when I go higher. It's burning one led on the next fan.

Is it because I have hooked the data up for all the fans in serie? Data out first fan data in next fan.

Legion2, 2020-08-06:
Ok, then please update the library to 0.14.1 and use the following scaling code instead, which scales each fan individual:
```cpp
	CLP::SegmentScaling segments[] = {{12, 15}, {12, 15}, {12, 15}};
	CLP::scaleSegments(&ledController, 1, segments, 3);
```

Justdigit, 2020-08-06:
Thanks mate.... Tried several different fans and values. But nothing takes a way that first pixel on fan...

I have made a video. https://photos.app.goo.gl/yeeJSPNQQNYtexwq5

This is what happen when I use this code.

Legion2, 2020-08-06:
Try this code:
```cpp
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2

CRGB ledsChannel1[45];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 45);
	ledController.addLEDs(0, ledsChannel1, 36);
	ledController.onUpdateHook(0, {
		CLP::SegmentScaling segments[] = {{12, 15}, {12, 15}, {12, 15}};
		CLP::scaleSegments(&ledController, 0, segments, 3);
	});
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}
```

Justdigit, 2020-08-06:
> Try this code:
> 
> ```c++
> #include <CorsairLightingProtocol.h>
> #include <FastLED.h>
> 
> #define DATA_PIN_CHANNEL_1 2
> 
> CRGB ledsChannel1[45];
> 
> CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
> FastLEDController ledController(true);
> CorsairLightingProtocolController cLP(&ledController, &firmware);
> CorsairLightingProtocolHID cHID(&cLP);
> 
> void setup() {
> 	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 45);
> 	ledController.addLEDs(0, ledsChannel1, 36);
> 	ledController.onUpdateHook(0, {
> 		CLP::SegmentScaling segments[] = {{12, 15}, {12, 15}, {12, 15}};
> 		CLP::scaleSegments(&ledController, 0, segments, 3);
> 	});
> }
> 
> void loop() {
> 	cHID.update();
> 
> 	if (ledController.updateLEDs()) {
> 		FastLED.show();
> 	}
> }
> ```

I get a void loop error when compiling??

Legion2, 2020-08-06:
There was a mistake in the code I copied from you, I have now updated the code.

Justdigit, 2020-08-06:
> There was a mistake in the code I copied from you, I have now updated the code.

You are the man.!! Its working!!  Many thanks and for your patience.

Legion2, 2020-08-06:
> So i cant you use a second channel on this Arduino?

you can use it, I just removed it, to make the code more clear. You can also add more fans and other stuff.

closed 2020-08-06 by Legion2

Justdigit, 2020-08-06:
> > So i cant you use a second channel on this Arduino?
> 
> you can use it, I just removed it, to make the code more clear. You can also add more fans and other stuff.

Yes i already put that piece back and its working..... no the wait is for some ledstrips....awesome project!!!