Legion2/CorsairLightingProtocol #167: CorsairLightingProtocolHID' does not name a type; did you mean 'CorsairLightingProtocolSerial'?

TheAdminFrmoHell, 2020-08-10:
**Describe the problem**
Error compiling

**Screenshots**
https://pastebin.com/q8vEX2pD?fbclid=IwAR01ss11M-yhlZZCl9bquZ5z7j9MZdZ3WyDCoHGZBGMJp5IDL9gQ52kQUTk

**System (please complete the following information):**
Windows 10
 - Board: Pro Micro
 - IDE: Arduino IDE
 - Version: 1.8.13
 - Sketch: All of the example files

**Code changes**
Tried with both the original file and a file with changes.

Legion2, 2020-08-10:
[image]
Did you select the CLP board?

TheAdminFrmoHell, 2020-08-10:
I was just coming back here to write it was an above keyboard failure 😂
I had selected the Pro mini/micro, I've changed it to CLP and now it works perfectly

closed 2020-08-10 by Legion2